adafruit/circuitpython #6553: Pins C14, C15 "in use" right from the start on STM32

rgbpcb, 2022-07-04:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-31-gc1b00442a on 2022-07-02; stm32f411ce-blackpill with STM32F411CE
```

### Code/REPL

```python
import board, digitalio
digitalio.DigitalInOut(board.C14)
```

### Behavior

`ValueError: C14 in use`

### Additional information

If I comment out the calls to `never_reset_pin_number()` [here](https://github.com/adafruit/circuitpython/blob/main/ports/stm/peripherals/stm32f4/stm32f411xe/gpio.c#L43-L44), then the error doesn't appear anymore. But then using those pins among the `row_pins` of `keypad.KeyMatrix()` prevents the key matrix from detecting any events, also from other pins. The condition `if key_matrix.events.get()` (where `key_matrix` was the output of `keypad.KeyMatrix()`) is never triggered. Without using pins C14 and C15, it works.

rgbpcb, 2022-07-05:
On the `stm32f411ce-blackpill` board, pins C14 and C15 are connected to the low-frequency crystal oscillator. However, how does CircuitPython know about that, even though https://github.com/adafruit/circuitpython/blob/main/ports/stm/boards/stm32f411ce_blackpill/mpconfigboard.h#L37 says `#define BOARD_HAS_LOW_SPEED_CRYSTAL (0)` (which as far as I understand means that CircuitPython will assume that there is _no_ low-speed crystal)?

And since `#define BOARD_HAS_LOW_SPEED_CRYSTAL (0)` doesn't make pins C14 and C15 available, what does? (Asking for boards without a low-speed crystal.)

dhalbert, 2022-07-05:
See comments in https://github.com/adafruit/circuitpython/pull/2361 for some background on pin stuff on this board.

rgbpcb, 2022-07-05:
Due to the problems with `keypad`, I'll first try something simpler (like `digitalio` I guess).

rgbpcb, 2022-07-14:
The pins C14 and C15 work both in the input and output direction with `digitalio` (on boards where there's no crystal at those pins).

But as described above, using those pins in `keypad` causes the entire keypad to not detect any events, even for all other pins.

What might be causing `keypad` to fail even if `digitalio` works? Does CircuitPython associate some additional properites/variables to pins, causing pins C14 and C15 to behave differently from other pins in the context of `keypad`?

tannewt, 2022-07-14:
Keypad.KeyMatrix uses the digitalio code internally so I'd expect it to work: https://github.com/adafruit/circuitpython/blob/main/shared-module/keypad/KeyMatrix.c#L56= Must be some quirk with those pins.

rgbpcb, 2022-07-14:
What kind of quirk for example? How can this be fixed? Note that using pins C14 and C15 in `keypad.KeyMatrix()` stops `keypad` from working for other pins as well. So maybe `keypad` uses some variables/constants that are associated with pins C14 and C15 while setting up some global settings. What variables/constants are associated with pins?

What apart from `never_reset_pin_number()` (which I commented out [here](https://github.com/adafruit/circuitpython/blob/main/ports/stm/peripherals/stm32f4/stm32f411xe/gpio.c#L43-L44) in order to get `digitalio.DigitalInOut(board.C14)` to work) and `BOARD_HAS_LOW_SPEED_CRYSTAL` (which is `0` already [here](https://github.com/adafruit/circuitpython/blob/main/ports/stm/boards/stm32f411ce_blackpill/mpconfigboard.h#L37)) affects the variables/constants associated with those pins?

Maybe related: If `BOARD_HAS_LOW_SPEED_CRYSTAL` were `1`, how would CircuitPython know _at which pins_ it should expect a low-speed crystal?

I might not manage to meaningfully modify the code of `keypad`, but does `keypad` already offer some kind of verbose logging in order to better understand why it thinks that none of the keys are pressed? Can I somehow ask `keypad` what it thinks the state of certain keys in the matrix is?

rgbpcb, 2022-07-14:
The summary so far:

- For pins C14 and C15 to be usable by `digitalio` and `keypad`, `never_reset_pin_number()` should be commented out at https://github.com/adafruit/circuitpython/blob/main/ports/stm/peripherals/stm32f4/stm32f411xe/gpio.c#L43-L44
- You might want to actually comment it out there in your main codebase. (Should the ticket remain open until then?) I haven't tested how the low-speed crystal on the stm32f411ce-blackpill board connected to those pins interferes with their GPIO behavior. But those pins are broken out, so using them should cause a warning rather than an error.
- Why is `BOARD_HAS_LOW_SPEED_CRYSTAL` set to `0` in https://github.com/adafruit/circuitpython/blob/main/ports/stm/boards/stm32f411ce_blackpill/mpconfigboard.h#L37 ? Versions 2.0 and 3.1 of the board do seem to have a low-speed crystal.
- If `BOARD_HAS_LOW_SPEED_CRYSTAL` is `1`, how to tell CircuitPython at which pins to expect a low-speed crystal?
- `keypad.KeyMatrix()` seems to have a bug: it doesn't detect any events if `row_pins` are more than 9 pins.

tannewt, 2022-07-15:
> What kind of quirk for example? How can this be fixed? Note that using pins C14 and C15 in `keypad.KeyMatrix()` stops `keypad` from working for other pins as well. So maybe `keypad` uses some [...]

I was thinking maybe IOMUX changes for those pins since they also host a crystal. I wouldn't know specifics without looking into it. Generally there are pin structs for each pin. On STM I think there are also tables that list what pins can do certain functions.

> Maybe related: If `BOARD_HAS_LOW_SPEED_CRYSTAL` were `1`, how would CircuitPython know _at which pins_ it should expect a low-speed crystal?

Generally an MCU only has one location for a low speed crystal. (If it has multiple locations, then may CP only uses one position.) Most pins can't be used for it.

> I might not manage to meaningfully modify the code of `keypad`, but does `keypad` already offer some kind of verbose logging in order to better understand why it thinks that none of the keys are pressed? Can I somehow ask `keypad` what it thinks the state of certain keys in the matrix is?

I'm not sure. I haven't done keymatrix work myself. You could build CP yourself and add print statements into it.