adafruit/circuitpython #6554: Pin A10 "in use" right from the start on STM32

rgbpcb, 2022-07-04:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-31-gc1b00442a on 2022-07-02; stm32f411ce-blackpill with STM32F411CE
```

### Code/REPL

```python
import board, digitalio
digitalio.DigitalInOut(board.A10)
```

### Behavior

`ValueError: A10 in use`

### Additional information

[This code comment](https://github.com/adafruit/circuitpython/blob/main/ports/stm/supervisor/usb.c#L81) says that pin A10 gets occupied by USB_OTG_FS_ID. How do I prevent precisely that? Adding `#define BOARD_NO_USB_OTG_ID_SENSE (1)` to the board-specific `mpconfigboard.h` such as [ports/stm/boards/stm32f411ce_blackpill/mpconfigboard.h](https://github.com/adafruit/circuitpython/blob/main/ports/stm/boards/stm32f411ce_blackpill/mpconfigboard.h) causes the device to appear as "Unknown USB Device (Device Descriptor Request Failed)" in Device Manager in Windows 10, without REPL connection, without CIRCUITPY drive.

rgbpcb, 2022-07-05:
Here's the schematic: https://github.com/WeActTC/MiniSTM32F4x1/blob/master/HDK/MiniF4x1Cx_V31.pdf
Pin A10 of the MCU is connected to the pin header and to nothing else. How to avoid the immediate `ValueError: A10 in use`?

dhalbert, 2022-07-05:
See https://github.com/adafruit/circuitpython/pull/2361 for some comments on this board.

rgbpcb, 2022-07-14:
Should I comment out both `never_reset_pin_number(0, 10);` and `claim_pin(0, 10);` at https://github.com/adafruit/circuitpython/blob/main/ports/stm/supervisor/usb.c#L125-L126 ?

What is meant by "reset pin number" (reset from what value to what value and why?) and "claim pin" (claim in the sense that the user can't use that pin?) in those function names?

rgbpcb, 2022-07-14:
Commenting out `never_reset_pin_number(0, 10); claim_pin(0, 10);` or only `never_reset_pin_number(0, 10);` at https://github.com/adafruit/circuitpython/blob/main/ports/stm/supervisor/usb.c#L125-L126 causes the device to appear as "Unknown USB Device (Device Descriptor Request Failed)" in Device Manager in Windows 10, without REPL connection, without CIRCUITPY drive.

Why is that? How does `never_reset_pin_number(0, 10);` affect things that happen at that pin _before_ my code gets executed? What does `never_reset_pin_number` do? Or what else should I try?

tannewt, 2022-07-15:
> What is meant by ["reset pin number"](https://github.com/adafruit/circuitpython/blob/main/ports/stm/common-hal/microcontroller/Pin.c#L79) (reset from what value to what value and why?) and ["claim [...]

reset and claim are there to ensure one object is using a pin at a time. claim takes control of a pin and reset releases it back into the default state (sometimes floating, sometimes pulled).

> What does `never_reset_pin_number` do?

Never reset ensures that pins are untouched when most pins are reset after a VM run. We generally reset things before a VM run so that the next run starts from a known state. never_reset prevents that pin from being reset.

jepler, 2022-07-15:
On STM32F411, pin A10 has "USB_FS_ID" as an alternate function. The reference manual suggests that a bit in the hardware register OTG_FS_GUSBCFG can override the logic level present on the pin. 
[image]
[image]

rgbpcb, 2022-07-15:
> a bit in the hardware register OTG_FS_GUSBCFG can override the logic level present on the pin

Do you mean that that might create some logic level that makes USB work, and the aforementioned _resetting of pins_ might mess up that logic level during the first second(s) of bootup, and that's why "never reset" is important for USB to work?

rgbpcb, 2022-07-16:
When I do [`never_reset_pins[pin_port] |= 1 << pin_number;`](https://github.com/adafruit/circuitpython/blob/main/ports/stm/common-hal/microcontroller/Pin.c#L83) without doing [`claimed_pins[pin_port] |= 1 << pin_number;`](https://github.com/adafruit/circuitpython/blob/main/ports/stm/common-hal/microcontroller/Pin.c#L84-L85), then USB does work, but I get `ValueError: A10 in use`.

If I additionally comment out [`claimed_pins[i] = never_reset_pins[i];`](https://github.com/adafruit/circuitpython/blob/main/ports/stm/common-hal/microcontroller/Pin.c#L57), then I can use that pin.

I can do `import board, digitalio; digitalio.DigitalInOut(board.A10); print('foo')` and I see the output of the `print` statement in the REPL. But a second later, USB disconnects.

Next, I'll try to keep that pin at levels (using a pullup resistor or programming it) that prevent USB from disconnecting.

dhalbert, 2022-07-17:
"Never reset" means that the pin is not reset when the CircuitPython VM restarts, such as when you type ctrl-D. This is usually for pins whose state must be maintained across VM instantiations, such as display control pins, USB, SPI flash pins, etc. So if you disable part of the "never reset" mechanism, things may break.

Given the special USB use of `A10`, we might have been better off omitting it from `board` (that goes for C14 and C15 too). As I mentioned in a related issue, PR #2361, the PR that added this board, discusses a number of peculiarities of the blackpill. You might consider using a less problematic board that has sufficient GPIO's.

rgbpcb, 2022-07-17:
Is there a way in `digitalio` or `keypad` or other libraries to initialize the pin immediately with a certain `direction`, `pull`, and/or `value` instead of initializing with the default ones as in `digitalio.DigitalInOut(board.A10)` and changing them after that? Only by modifying `DigitalInOut.c` myself?

Does `digitalio.DigitalInOut(board.A10)` (and `HAL_GPIO_Init()` in [`stm32f4xx_hal_gpio.c`](https://github.com/adafruit/stm32f4xx_hal_driver/blob/master/Src/stm32f4xx_hal_gpio.c)) merely affect the level at that pin exactly as I want, or does it do some logic like "this pin can have certain USB functionality sometimes, therefore let's perform some USB-related special actions that affect other pins"?

> Given the special USB use of `A10`, we might have been better off omitting it from `board`

There are boards that successfully use pin A10 for GPIO with different firmware. I'm trying to get that to work with CircuitPython.

> (that goes for C14 and C15 too)

Please don't omit them. I got them to work already, I recently explained that bugfix in #6553. In a nutshell, CircuitPython currently claims them (`never_reset_pin_number()` claims them) **for all stm32f411xe MCUs** in https://github.com/adafruit/circuitpython/blob/main/ports/stm/peripherals/stm32f4/stm32f411xe/gpio.c#L43-L44 , but not all boards with these MCUs have those pins occupied by a low-speed crystal. So always claiming them is a bug. By commenting out those lines, I got pins C14 and C15 to work for `digitalio`.

> PR #2361, the PR that added this board, discusses a number of peculiarities of the blackpill

I'll have a look which peculiarities are BlackPill-specific as opposed to more generally related to stm32f411xe. I'm using (and sometimes modifying) the code you wrote for BlackPill, but I'm experimenting also with other schematics with the same MCU. In fact, I might also try the workaround from my previous post on the BlackPill itself.

dhalbert, 2022-07-18:
The way to manipulate the pins or board-specific setup at hard startup is in `board_init()`, in `board.c` for the board. There you could set the register mentioned [above](https://github.com/adafruit/circuitpython/issues/6554#issuecomment-1185913479) so that A10 was not treated specially.

dhalbert, 2022-07-18:
You want to force device mode, not host mode (host mode is what the host computer uses).

That is not a variable, it's just a constant you need to use to set the appropriate I/O register. An I/O register is in the address space, but is not regular memory. It controls an on-chip peripheral.

You would do something like this (code taken from https://github.com/hathach/tinyusb/blob/ecb899408bb95c5c0d50ebe2d887cd746104c534/hw/bsp/stm32f7/family.c#L257-L259):
```c
  // Force device mode
  USB_OTG_HS->GUSBCFG &= ~USB_OTG_GUSBCFG_FHMOD;
  USB_OTG_HS->GUSBCFG |= USB_OTG_GUSBCFG_FDMOD;
```

dhalbert, 2022-07-18:
The code above is for F7's or H7's, so sorry, yes, `FS`, not HS`. I didn't realize the register bit was already potentially set in `usb.c`.

I don't know why the current code flow is not working properly if you set `BOARD_NO_USB_OTG_ID_SENSE` to 1. If you put the code in `board_init()`, it happens really, really early. You could try it there and see if it works.

dhalbert, 2022-07-18:
you need to add the right `#include` files to get the definitions you need. `#define` is a C preprocessor setting. It is not C code that runs.

rgbpcb, 2022-07-18:
`supervisor/usb.c`, which uses `USB_OTG_FS`, has the following `#includes`:
```
#include "supervisor/usb.h"
#include "shared/runtime/interrupt_char.h"
#include "shared/readline/readline.h"
#include "lib/tinyusb/src/device/usbd.h"

#include "py/mpconfig.h"

#include "common-hal/microcontroller/Pin.h"
```
Any hints which ones I need for `board.c`? Otherwise I'll try using all.

rgbpcb, 2022-07-18:
I tried all those includes at once, USB works with REPL and CIRCUITPY, but doing `digitalio.DigitalInOut(board.A10)` disconnects the USB. `board.c` contains:
```c
#include "supervisor/usb.h"
#include "shared/runtime/interrupt_char.h"
#include "shared/readline/readline.h"
#include "lib/tinyusb/src/device/usbd.h"

#include "py/mpconfig.h"

#include "common-hal/microcontroller/Pin.h"

void board_init(void) {
    USB_OTG_FS->GUSBCFG &= ~USB_OTG_GUSBCFG_FHMOD;
    USB_OTG_FS->GUSBCFG |= USB_OTG_GUSBCFG_FDMOD;
}
```

dhalbert, 2022-07-18:
At this point I don't understand why it's not working the way the datasheet implies. You mentioned that there is other firmware that allows use of A10, so it's probably worth studying those to see how they get around this problem.

rgbpcb, 2022-07-18:
I agree, the behavior seems to contradict the datasheet.

As far as I understand, the `FDMOD` bit should be `1` while I do `digitalio.DigitalInOut(board.A10)`. Maybe some other code overwrote that bit after we had set it to `1` initially? How can I set it to `1` as **late** as possible, for example from a CircuitPython script?

dhalbert, 2022-07-19:
You can't set it from CircuitPython, and by `code.py` time it would be too late. `boot.py` might work but something might reset it in the interim. It's possible it's being reset somehow.

This kind of hair-pulling frustration is why we are not so fond of this board and its idea of exposing absolutely every pin. The A9 problem mentioned in #2361 was similar. Yes, A10 can be made available, but it's a lot of trouble, and causes a lot of support issues. Better to just hide it on a general-purpose board would be our philosophy.

dhalbert, 2022-10-05:
Closing this for now.

closed 2022-10-05 by dhalbert

rgbpcb, 2022-11-03:
> we are not so fond of this board and its idea of exposing absolutely every pin

Where in the documentation of an MCU ([example link](https://www.st.com/en/microcontrollers-microprocessors/stm32f411.html#documentation)) do I find the info which pins shouldn't be exposed/used?

For example, in Table 10 of [the datasheet of the aforementioned MCU](https://www.st.com/resource/en/datasheet/stm32f411ce.pdf), should I just avoid all pins mentioned in the column `OTG1_FS`, and all other pins are safe to use? Anything else apart from `OTG1_FS` to look out for in other MCUs?