adafruit/circuitpython #6556: audiomp3 on Qt Py ESP32-S3

scarolan, 2022-07-05:
I think it has enough horsepower and memory to run audiomp3...if you can add this to the built in modules for the Qt Py ESP32-S3 firmware, that would be awesome!

tannewt, 2022-07-05:
IIRC, this isn't easy because the MP3 decoder library we use doesn't have xtensa assembly like it does for ARM. @jepler would be the person who looked into this.

timonsku, 2023-11-29:
On the flow3r badge we use https://github.com/lieff/minimp3 in combination with micropython which gave us decent performance on the ESP32-S3 without any optimized assembly needed. Compiled with -O2.

Licensing is a bit mixed in the firmware but if you want a reference:
https://git.flow3r.garden/flow3r/flow3r-firmware/-/tree/main/components?ref_type=heads

scarolan, 2024-01-28:
amazing!

timonsku, 2024-01-28:
is this actually resolved though?

kollibar, 2024-03-25:
@jepler, do you think mp3 decoding will be add to esp-s3 board port ?
thanks

BrainBoardz, 2024-04-10:
I'm hoping that this lack of MP3 support in CP for the ESP32S3 can be sorted out in a future release of CP.  It's very unfortunate that CP supports the S3 so well overall, and yet there is this gaping hole regarding audiomp3 support. MP3 playback is used for so many things (gaming, general audio, AI applications etc.) that not having this does limit the value of CP with respect to using it  with ESP32s.  I understand this is due to the lack of an xtensa assembly, so that this is just the way it will always be,  and it likely cannot ever be resolved.  But, if there is any way this could get sorted out  it would be a very welcome addition.

dhalbert, 2024-04-10:
It's not that it can't be resolved, but we have to find a new implementation of the MP3 decoder. minimp3 appears to be Creative Commons CC0, though the flow3r licensing is GPL, it appears.

timonsku, 2024-04-10:
> It's not that it can't be resolved, but we have to find a new implementation of the MP3 decoder. minimp3 appears to be Creative Commons CC0, though the flow3r licensing is GPL, it appears.

We did not do much here other than a couple convenience functions. We were forced into GPL unfortunately due to other dependencies and not wanting to overcomplicate things. Not sure how much of what we wrote for the Python side would be directly usable for CPY anyway.

Though I have some time open up atm, if you could use some help on that end could ask Limor if that might be something worth while to do for me?

BrainBoardz, 2024-04-10:
Hi dhalbert and timonsku: 

Thanks for the update. So that means that there is at least some potential for this to be resolved in the future if the technical
challenges of implementing minimp3 could be worked out , but that the a more  serious roadblock is licensing constraints. I guess this cant just be reverse engineered, Halt and Catch Fire style, ha ha. Would be amazing if it made it onto a todo list.  The CP support for S3 works great. So close to perfection...

dhalbert, 2024-04-10:
minimp3 is Creative Commons CC0. That is even more permissive than the MIT license we use.

dhalbert, 2026-09-13:
Fixed by #9218, a long time ago.

closed 2026-09-13 by dhalbert