adafruit/circuitpython #6559: wifi.radio.tx_power does not support float

askpatrickw, 2022-07-05:
### CircuitPython version

```python
The [Wemos docs say to use 8.5db](https://www.wemos.cc/en/latest/c3/c3_mini.html#about-wifi) to work around the WIFI connect issue.

## Workaround

If you use 8 or 9 you can still connect.
```

### Code/REPL

```python
>>> import wifi
>>> wifi.radio.tx_power = 8.5
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert float to int
>>> wifi.radio.tx_power = 8
>>> wifi.radio.connect(SSID, PASSWD)
>>> wifi.radio.ipv4_address
192.168.1.53
```

### Behavior

TypeError: can't convert float to int

askpatrickw, 2022-07-05:
Related to PR #6542

askpatrickw, 2022-07-08:
@todbot any chance you want to look at this one?

todbot, 2022-07-08:
This was by design. The underlying API is int so I figured this attribute should be that way too.  Is that a bad assumption?

todbot, 2022-07-08:
Like, what do they mean exactly "set to 8.5dB" when this is the API : https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t

dhalbert, 2022-07-08:
The units are 0.25dBm, but the actual power setting is coarser than that, not even in 1dBm units:
https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s2/api-reference/network/esp_wifi.html?highlight=esp_wifi_set_max_tx_power#_CPPv425esp_wifi_set_max_tx_power6int8_t

> [esp_err_t](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s2/api-reference/system/esp_err.html#_CPPv49esp_err_t)esp_wifi_set_max_tx_power(int8_t power)
> Set maximum transmitting power after WiFi start.
> 
> Attention
> 1. Maximum power before wifi startup is limited by PHY init data bin.
> 
> Attention
> 2. The value set by this API will be mapped to the max_tx_power of the structure [wifi_country_t](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s2/api-reference/network/esp_wifi.html?highlight=esp_wifi_set_max_tx_power#structwifi__country__t) variable.
> 
> Attention
> 3. Mapping Table {Power, max_tx_power} = {{8, 2}, {20, 5}, {28, 7}, {34, 8}, {44, 11}, {52, 13}, {56, 14}, {60, 15}, {66, 16}, {72, 18}, {80, 20}}.
> 
> Attention
> 4. Param power unit is 0.25dBm, range is [8, 84] corresponding to 2dBm - 20dBm.
> 
> Attention
> 5. Relationship between set value and actual value. As follows: {set value range, actual value} = {{[8, 19],8}, {[20, 27],20}, {[28, 33],28}, {[34, 43],34}, {[44, 51],44}, {[52, 55],52}, {[56, 59],56}, {[60, 65],60}, {[66, 71],66}, {[72, 79],72}, {[80, 84],80}}.
> 
> Return
> ESP_OK: succeed
> 
> ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
> 
> ESP_ERR_WIFI_NOT_START: WiFi is not started by esp_wifi_start
> 
> ESP_ERR_WIFI_ARG: invalid argument, e.g. parameter is out of range
> 
> Parameters
> power: Maximum WiFi transmitting power.

dhalbert, 2022-07-08:
I don't know what's going on internally - that code is in a binary blob. It could be something about how the radio hardware works. But in any case there's significant quantization. I too am puzzled about the Wemos instructions.

Have either of you tried wifi at various power settings to see if some don't work at all?

todbot, 2022-07-08:
They worked at all the 6 or so settings I tested, showing corresponding changes in TX power using a WiFi sniffer. 
This is not a big change, it's just frustrating because Patrick is right this should be a float. :)

dhalbert, 2022-07-08:
I agree, a float would be good. No issue changing it.

todbot, 2022-07-08:
okay, PR incoming shortly

commit c56f363a (2022-07-11): Merge pull request #6572 from todbot/wifi-tx-power-float

Fix #6559
commit d3e1d1b1 (2022-07-11): Fix #6559
closed 2022-07-11 by tannewt