adafruit/circuitpython #6563: UF2 images built for STM32F405 boards are incompatible with TinyUF2

NathanY3G, 2022-07-06:
### CircuitPython version

```python
8.0.0-alpha.1
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

### Description

I believe I have unwillingly discovered an incompatibility between the UF2 images built for STM32F405 boards and TinyUF2. Images produced both locally and downloaded from http://circuitpython.org cannot be flashed using TinyUF2.

This appears to be due to the `family ID` contained within the UF2 images being different from TinyUF2's expectation. Circuit Python builds use a value of `0x6d0922fa` (STM32F407) whilst TinyUF2 is expecting `0x57755A57` (STM32F4).

tannewt, 2022-07-06:
@NathanY3G Is the 407 family correct? Are you interesting in updating the Makefile to fix this?

NathanY3G, 2022-07-07:
Hi @tannewt. I don't believe the use of the F407 family ID is correct. Looking at the [STM32F4 portfolio](https://www.st.com/content/ccc/fragment/product_related/series_information/series_level_diagram/group0/6e/ab/f8/ba/9c/9a/4d/37/obn_stm32f4_series_ss1577/files/obn_stm32f4_series_ss1577.jpg/jcr:content/translations/en.obn_stm32f4_series_ss1577.jpg) on ST's web-site, the F405s and F407s are two different product lines within the STM32F4 series.

As always, I am more than happy to try and fix issues! I will take a look.

hathach, 2022-07-08:
this is a bit ambiguous. according to https://github.com/microsoft/uf2/blob/master/utils/uf2families.json There is defined for stm32f407 and stm32f4 (also f407vg but I guess we could ignore this)

https://github.com/microsoft/uf2/blob/ffe793f973ff8fc288365a2183c9727aa40aed07/utils/uf2families.json#L78-L80

https://github.com/microsoft/uf2/blob/ffe793f973ff8fc288365a2183c9727aa40aed07/utils/uf2families.json#L118-L120

My guess is the f407 was added first, then the family F4 was added later. The ideal solution is having all F4 shared the same family ID much like the rest (F7, H7, L4). Not sure if it would have impact on existing users. We would also change tinyuf2 to match circuitpython as well if needed

NathanY3G, 2022-07-08:
I agree. It also looks like there used to be a lot more family IDs for the ST processors before a refactoring to a single source of truth: https://github.com/microsoft/uf2/pull/47.

I was considering raising an issue to clarify/improve the situation unless you would like to do this?

NathanY3G, 2022-07-10:
I have a create a PR to make the UF2 family IDs consistent with TinyUF2 for STM32F405 boards. However, this is currently marked as draft as I wish to test my changes further.

hathach, 2022-07-17:
@NathanY3G should this be fixed by #6577 and ok to close now ?

NathanY3G, 2022-07-18:
Thanks for the reminder @hathach. I will close this issue :grinning:

closed 2022-07-18 by NathanY3G