adafruit/circuitpython #6573: ItsyBitsy M0 Express crash with Hardfault Handler error

heliosoph, 2022-07-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-09; Adafruit ItsyBitsy M0 Express with samd21g18
Board ID:itsybitsy_m0_express

Same behaviour with

Adafruit CircuitPython 7.3.1 on 2022-06-22; Adafruit ItsyBitsy M0 Express with samd21g18
Board ID:itsybitsy_m0_express

Both are the German versions
```

### Code/REPL

```python
file code.py:
import test

file lib/test.py:
import time
import board
import digitalio
from audiocore import RawSample
# from adafruit_mcp230xx.mcp23017 import MCP23017
```

### Behavior

This is the complete REPL output:

", Zeile <stdin>
>>> 
weicher reboot

Automatisches Neuladen ist deaktiviert.
Sicherheitsmodus aktiv! Gespeicherter Code wird nicht ausgeführt

Du bist im abgesicherten Modus weil:
Der CircuitPython-Kerncode ist hart abgestürzt. Hoppla!
Absturz in den HardFault_Handler.
Bitte melden Sie ein Problem mit dem Inhalt Ihres CIRCUITPY-Laufwerks unter
https://github.com/adafruit/circuitpython/issues

Drücke eine beliebige Taste um REPL zu betreten. Drücke STRG-D zum neuladen.

### Description

I have a ItsyBitsy M0 Express with Bootloader 3.14.0
I do the following steps:
First I erase the filesystem with
`import storage`
`storage.erase_filesystem`
from the REPL console.
Then I copy the libraries adafruit_bus_device and adafruit_mcp230xx from the 7.x or 8.x bundle release 20220708 into the lib folder, depending on the CP version I use for the test.
Then I copy the two files code.py and lib/test.py (see above) into the CIRCUITPY drive.
Now the system runs.
Then, in the lib/test.py file I uncomment the last line.
Now the system crashes and gives the output shown above.
You can start again and repeat by erasing the file system again.
This works both with CP 7.3.1 and CP 8.0.0 alpha 1

### Additional information

The board is standalone. Same behaviour with a MCP23017 connected.
I have opened a thread in the adafruit forums:
[https://forums.adafruit.com/viewtopic.php?f=62&t=192583&p=931844#p931844](url)

DavePutz, 2022-08-27:
This looks like a stack size issue. Running 8.0-beta-1 I get:
code.py output:
Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "test.py", line 5, in <module>
  File "adafruit_mcp230xx/mcp23017.py", line 18, in <module>
  File "adafruit_mcp230xx/mcp230xx.py", line 16, in <module>
  File "adafruit_mcp230xx/mcp23xxx.py", line 14, in <module>
RuntimeError: maximum recursion depth exceeded

Building with CIRCUITPY_DEFAULT_STACK_SIZE set to 4608 (from 3584) allows code.py to finish successfully.

dhalbert, 2022-09-21:
Thanks @DavePutz for diagnosis. The stack size on SAMD21 boards is very limited due to the limited ram (32kB). And if we increase the stack size, then some programs don't run because they run out of heap memory.

I incorporated the program into a single code.py and avoided the error. You can also put
```py
import supervisor
supervisor.set_next_stack_limit(<4608 or larger)
```
in `boot.py`. This will cause `code.py` to run with the specified stack size.

See https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.set_next_stack_limit

closed 2022-09-21 by dhalbert