adafruit/circuitpython #6593: Willing to port Pico W MicroPython code to CircuitPython

richsad, 2022-07-14:
Don't know where to leave this. I'm a 40 veteren and have shipped two commercial dev platforms (Symantec C++, Visual Cafe (also originally Symantec). More relevant, I'm polyglot programmer of libraries going back to C/C++ standard libraries. I've also worked on OSes (3). I am a busy CTO but have been looking for something important to do for CP. Check out https://www.linkedin.com/in/richsadowsky/

and write to me at rich@waverleysoftware.com. That reaches me fastest

askpatrickw, 2022-07-14:
@richsad  Discord is the place to be ... checkout the circuitpython-dev channel.

http://adafru.it/discord

tannewt, 2022-07-14:
Thanks for the offer! Closing in favor of #6558

closed 2022-07-14 by tannewt