adafruit/circuitpython #6597: Failure to run on QT Py ESP32-C3

archaic0, 2022-07-14:
### CircuitPython version

```python
7.3.1 / 8.0
```

### Code/REPL

```python
python -m esptool --chip esp32c3 --port COM30 --baud 115200 --before default_reset --after hard_reset --no-stub write_flash 0x0 c:\temp\adafruit-circuitpython-adafruit_qtpy_esp32c3-en_US-7.3.1.bin
```

### Behavior

After install and reboot, I get a welcome message in Thonny in the Shell output window exactly one time, but no interactive >> prompt in the shell.  Also, no USB drive is mounted.  (May also be a Thonny issue in lacking support for the C3, but that doesn't explain why no USB drive is mounted)

Welcome banner says: Adafruit CircuitPython 7.3.1 on 2022-06-22; Adafruit QT Py ESP32 Adafruit 
CircuitPython 7.3.1 on 2022-06-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4

After that first time, every attempt to connect to the device results in the neopixel flashing 4 yellow, 1 green, then flashing 1 green every 5 seconds.

Installing v8 produces a different welcome message, but same behavior: (Ctrl-C and Ctrl-D do nothing, again, no >> prompt)
Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-09; Adafruit QT Py ESP32C3 with ESP32-C3FN4
Board ID:adafruit_qtpy_esp32c3

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press Ctrl-C to enter the REPL. Use CTRL-D to reload.

Installing latest build from S3 produces garbage characters (those are in the real output, not typos here):
Adafruit CircuitPython 8.0.0-alpha.1-47-g9d5b04ebc on 2022-07-12
]0;(S) Wi-Fi: off\Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py]0;(S) Wi-Fi: off\ output:
Hello World!

Code done running.
]0;(S) Wi-Fi: off\
Press Ctrl-C to enter the REPL. Use CTRL-D to reload.

RetiredWizard, 2022-07-14:
The QTPY ESP32-C3  has built-in USB-to-Serial, but not native USB - it cannot act as a keyboard or disk drive. Microcontroller's without native USB have not gotten much attention from CircuitPython up until now because they couldn't mount the CIRCUITPY drive. However, now that the BLE and Web workflow is starting to be available these chips may get more development attention.

Your issue is probably the same as #6220

tannewt, 2022-07-14:
The weird characters are due to the new title bar updates. We've already got a bug to track fixing Thonny and other tools here: https://github.com/adafruit/circuitpython/issues/6548

closed 2022-07-14 by tannewt

todbot, 2022-07-29:
I am still having problems on QTPy ESP32C3, even after #6548 and #6666.  I built and flashed from main, and can reliably get a REPL now after a reset, but if I copy-n-paste this as a block from my editor into a terminal running `tio` to the REPL, it hangs:

```py
f = open('.env', 'w')
f.write("CIRCUITPY_WIFI_SSID='todbottestwifi'\n")
f.write("CIRCUITPY_WIFI_PASSWORD='sshh12345'\n")
f.write("CIRCUITPY_WEB_API_PASSWORD='passw0rd'\n")
f.close()
```

The hang isn't consistent where in the paste it happens, but usually after about 100 characters.  Disconnecting & reconnecting `tio` doesn't release the hang. 

But I can at least copy-n-paste line-by-line and I don't have dropped characters, which was not the case yesterday!

lundmar, 2022-07-30:
@todbot I know it has been reported before that some low power embedded targets would would loose characters when doing block pastes using tio because it used to write to the serial port one character at a time and that may have caused issues with some usb serial devices because of how the low power target can not keep up with the USB ring buffering. This issue has been resolved since tio v1.41. Now, if you paste a block tio will process the block and write it to the serial port as a block.

Would be great if you could test using latest tio version to confirm.

Also, it may not be tio related. It may simply be that the ESP32C3 target can not process the pasted lines as quickly as they come in and you may need line delays. tio recently added a output line delay feature to handle this exact issue (--output-line-delay).

RetiredWizard, 2022-07-30:
I think I'm seeing the same thing todbot is, except I'm using a puTTY terminal on Windows rather than tio.

lundmar, 2022-07-30:
Right, then it is probably because the device can't quite process the incoming lines as quickly as they come in. If so, using the tio output line delay feature with an appropriate delay value should make it possible to paste blocks.

tannewt, 2022-08-01:
@todbot Thanks for testing! Please file new issues. It's harder to track comments on closed issues. (We could reopen if the problem is the same as original.) I've opened #6678 for your report.