adafruit/circuitpython #6599: Attempt to build on A Beaglebone Black fails

scirelli, 2022-07-14:
### CircuitPython version

```python
main branch
```

### Code/REPL

```python
No code was run
```

### Behavior

Did not run any code.

### Description

I get an error when trying to install the requirements-dev.txt

> Encountered error while trying to install package.
> ╰─> RPi.GPIO
> ERROR: Couldn't install package: adafruit-circuitpython-typing

Logs below:
```
 Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
 Collecting adafruit-circuitpython-typing
   Using cached https://www.piwheels.org/simple/adafruit-circuitpython-typing/adafruit_circuitpython_typing-1.7.1-py3-none-any.whl (9.2 kB)
 Collecting adafruit-circuitpython-busdevice
   Using cached https://www.piwheels.org/simple/adafruit-circuitpython-busdevice/adafruit_circuitpython_busdevice-5.1.10-py3-none-any.whl (7.3 kB)
 Collecting Adafruit-Blinka>=7.0.0
   Using cached https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-8.0.2-py3-none-any.whl (231 kB)
 Collecting pyftdi>=0.40.0
   Using cached https://www.piwheels.org/simple/pyftdi/pyftdi-0.54.0-py3-none-any.whl (144 kB)
 Collecting RPi.GPIO
   Using cached RPi.GPIO-0.7.1.tar.gz (29 kB)
   Preparing metadata (setup.py): started
   Preparing metadata (setup.py): finished with status 'done'
 Requirement already satisfied: rpi-ws281x>=4.0.0 in ./.venv/lib/python3.10/site-packages (from Adafruit-Blinka>=7.0.0->adafruit-circuitpython-busdevice->adafruit-circuitpython-typing->-r /tmp/pipenv-exl2oo
v_-requirements/pipenv-dn4pjcjs-requirement.txt (line 1)) (4.3.4)
 Collecting Adafruit-PureIO>=1.1.7
   Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
 Collecting Adafruit-PlatformDetect>=3.13.0
   Using cached https://www.piwheels.org/simple/adafruit-platformdetect/Adafruit_PlatformDetect-3.25.0-py3-none-any.whl (18 kB)
 Requirement already satisfied: sysv-ipc>=1.1.0 in ./.venv/lib/python3.10/site-packages (from Adafruit-Blinka>=7.0.0->adafruit-circuitpython-busdevice->adafruit-circuitpython-typing->-r /tmp/pipenv-exl2oov_
-requirements/pipenv-dn4pjcjs-requirement.txt (line 1)) (1.1.0)
 Collecting pyserial>=3.0
   Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
 Collecting pyusb!=1.2.0,>=1.0.0
   Using cached https://www.piwheels.org/simple/pyusb/pyusb-1.2.1-py3-none-any.whl (58 kB)
 Building wheels for collected packages: RPi.GPIO
   Building wheel for RPi.GPIO (setup.py): started
   Building wheel for RPi.GPIO (setup.py): finished with status 'error'
   Running setup.py clean for RPi.GPIO
 Failed to build RPi.GPIO
 Installing collected packages: RPi.GPIO, pyserial, pyusb, Adafruit-PureIO, Adafruit-PlatformDetect, pyftdi, Adafruit-Blinka, adafruit-circuitpython-busdevice, adafruit-circuitpython-typing
   Running setup.py install for RPi.GPIO: started
   Running setup.py install for RPi.GPIO: finished with status 'error'
 error: subprocess-exited-with-error

   × python setup.py bdist_wheel did not run successfully.
   │ exit code: 1
   ╰─> [50 lines of output]
       running bdist_wheel
       running build
       running build_py
       creating build
       creating build/lib.linux-armv7l-cpython-310
       creating build/lib.linux-armv7l-cpython-310/RPi       copying RPi/__init__.py -> build/lib.linux-armv7l-cpython-310/RPi
[... 111 lines cut ...]
             if (!output())
                  ~~~~~~^
       source/py_gpio.c:475:18: error: called object type 'PyObject *' (aka 'struct _object *') is not a function or function pointer
             if (!output())
                  ~~~~~~^
       4 warnings and 5 errors generated.
       error: command '/usr/bin/clang' failed with exit code 1
       [end of output]

   note: This error originates from a subprocess, and is likely not a problem with pip.
 error: legacy-install-failure

 × Encountered error while trying to install package.
 ╰─> RPi.GPIO

 note: This is an issue with the package mentioned above, not pip.
 hint: See above for output from the failure.
ERROR: Couldn't install package: adafruit-circuitpython-typing
```

### Additional information

This is on a Beaglebone Black
Python 3.10.4
pip 22.1

scirelli, 2022-07-15:
I tried to do both actually. 
This issue is from me trying to build CircuitPython on the Beaglebone Black (BBB). I was following those instructions ([learn.adafruit.com/building-circuitpython](https://learn.adafruit.com/building-circuitpython)).

I honestly got a little confused, I started to follow the [Linux Setup](https://learn.adafruit.com/building-circuitpython/linux) thinking, I'm on the BBB, it's Linux, I'll follow these. But I think they are for setting up a cross compiler from something like an x86 Ubuntu Desktop? If I'm wrong I'll go back and try that again. I jumped ahead to [Build CircuitPython](https://learn.adafruit.com/building-circuitpython/build-circuitpython). I get the above error on the [Install Required Python Packages](https://learn.adafruit.com/building-circuitpython/build-circuitpython#install-required-python-packages-3086211) step.

tannewt, 2022-07-15:
Ya, I think the instructions are for Ubuntu. What port are you trying to build? You could skip the python install and just install what you need to fix failures when actually building. #circuitpython-dev on the [Discord](https://adafru.it/discord) server is a great place to get help with building CP.

scirelli, 2022-07-16:
I'm trying to build on the Beaglebone Black for the Beaglebone Black, it's a Sitara AM3358BZCZ100, 1GHZ, processor.

dhalbert, 2022-07-17:
I think the problem will be in finding an `arm-none-eabi-gcc` ross-compiler toolsuite built for the BBB.

dhalbert, 2022-10-05:
Closing this for now due to age.

closed 2022-10-05 by dhalbert