adafruit/circuitpython #6601: `keypad.KeyMatrix()` not detecting any events if `row_pins` are more than nine pins

rgbpcb, 2022-07-14:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-31-gc1b00442a on 2022-07-02; stm32f411ce-blackpill with STM32F411CE
```

### Code/REPL

```python
import board, keypad

key_matrix = keypad.KeyMatrix(
    row_pins=(board.A0, board.A1, board.A2, board.A5, board.A7, board.A8, board.B8, board.B9, board.C13, board.C14, board.C15, ),
    column_pins=(board.B14, board.B12, board.B10, board.B1, board.B0, board.A4, board.A3, ),
)

while True:
    event = key_matrix.events.get()
    if event:
        print(event)
```

### Behavior

Nothing gets printed when keys are pressed.

### Additional information

Removing a few pins from `row_pins` such that nine or fewer of them remain fixes the issue.

dhalbert, 2022-07-18:
I tested a similar program on an Adafruit Feather STM32F405. I don't have a keyboard matrix with that many rows to try to try, so I wired up ten tactile pusbuttons, using 10 pins for the row and 1 column pin. I was not able to reproduce the problem

Did you try this without A10, C14, or C15 at all?

```py
from board import *
import keypad

key_matrix = keypad.KeyMatrix(
    row_pins=(A1, A2, A3, A4, A5, SCK, MOSI, MISO, RX, TX),
    column_pins=(D12, ),
)

while True:
    event = key_matrix.events.get()
    if event:
        print(event)
```

rgbpcb, 2022-08-07:
> Did you try this without A10, C14, or C15 at all?

I'm not using pin A10. For pins C14 and C15, I'm using [this bugfix](https://github.com/adafruit/circuitpython/issues/6553) (i.e. commenting out `never_reset_pin_number()` [here](https://github.com/adafruit/circuitpython/blob/main/ports/stm/peripherals/stm32f4/stm32f411xe/gpio.c#L43-L44)) that gets them to work (except in the situations listed below). But you're right, leaving them out of `row_pins` and `column_pins` allows to detect _some_ events (I haven't yet checked whether _all_ possible events get detected).

The situations where pins C14 and C15 prevent `keypad.KeyMatrix()` from detecting any events (also events from other pins) are:

- The code mentioned [above](https://github.com/adafruit/circuitpython/issues/6601#issue-1305293036) (i.e. more than nine row pins, some of which are C14 and/or C15) (the "or" is new, determined by trying different variations of `row_pins`)
- Newly discovered situation: Like the code above, but swapped `row_pins` and `column_pins` (and thus also `columns_to_anodes=False`). At first, everything works fine, but then quickly pressing eight keys simultaneously stops `key_matrix` from detecting any further events.

I'm using a board which (unlike the BlackPill) doesn't have a crystal at those pins. In `mpconfigboard.h`, I have `#define BOARD_HAS_LOW_SPEED_CRYSTAL (0)` (which as far as I understand means that CircuitPython will assume that there is no low-speed crystal).

So why do pins C14 and C15 affect the behavior of other pins? Does CircuitPython give them some additional special treatment except for the `never_reset_pin_number()` that I commented out? Or does the MCU consider certain signals at those pins as signals coming from a crystal (despite `#define BOARD_HAS_LOW_SPEED_CRYSTAL (0)`), and partially stops working? (KeyMatrix events don't get detected, but other commands such as `print('foo')` still work though.)

dhalbert, 2023-10-28:
Maybe related to #6553

closed 2023-10-28 by dhalbert