adafruit/circuitpython #6602: audiobusio.PDMIn not working on XIAO BLE nRF52840 Sense

114514sht, 2022-07-15:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.1 on 2022-06-22; Seeed XIAO nRF52840 Sense with nRF52840
```

### Code/REPL

```python
import time
import array
import math
import board
import audiobusio

# Remove DC bias before computing RMS.
def mean(values):
    return sum(values) / len(values)

def normalized_rms(values):
    minbuf = int(mean(values))
    samples_sum = sum(
        float(sample - minbuf) * (sample - minbuf)
        for sample in values
    )

    return math.sqrt(samples_sum / len(values))

# Main program
mic = audiobusio.PDMIn(board.PDM_CLK, board.PDM_DATA, sample_rate=16000, bit_depth=16)
samples = array.array('H', [0] * 160)

while True:
    mic.record(samples, len(samples))
    magnitude = normalized_rms(samples)
    print((magnitude,))
    time.sleep(0.1)
```

### Behavior

-PDMIn.record() function writes constant 32760s to the buffer array
-not responding to any sound input from the PDM microphone

### Description

-might be relate to #5967

517687486, 2022-07-31:
我也遇到过和你一样的问题
I've had the same problem as you

这可能是麦克风电源没有打开导致的
This may be caused by the microphone power not turned on

你可以尝试加入以下代码
You can try adding the following code
```python
micpwr = digitalio.DigitalInOut(board.MIC_PWR)
micpwr.direction = digitalio.Direction.OUTPUT
micpwr.value = True
time.sleep(0.1)

mic = audiobusio.PDMIn(board.PDM_CLK,board.PDM_DATA,bit_depth=16)
```

希望对你有帮助
I hope that will be helpful.

114514sht, 2022-07-31:
问题解决了，谢谢。
Problem solved, thank you.

closed 2022-07-31 by 114514sht