adafruit/circuitpython #6609: Web serial input thrown off by left arrow

FoamyGuy, 2022-07-16:
### CircuitPython version

```python
Build from current main ESP32-S2 TFT.
```

### Code/REPL

```python
print("hi")
```

But with arrow keys used to move cursor after the parens.
```

### Behavior

The left arrow does move the cursor in the input box, but the text getting sent to serial doesn't come out the same, it seems to be missing the left arrow presses that changed the location of the cursor. 

### Additional information

https://user-images.githubusercontent.com/2406189/179326903-9564d0a2-2b0d-4d57-a9c6-ec011d02e9c1.mp4

FoamyGuy, 2022-07-16:
https://user-images.githubusercontent.com/2406189/179326903-9564d0a2-2b0d-4d57-a9c6-ec011d02e9c1.mp4

tannewt, 2022-07-25:
I'm not sure what the right thing to do here is. We could send extra characters to delete and send the rest of the line. I'm worried it's going to add a lot of logic. This might be a case that the full serial on code.circuitpython.org is better to use.

makermelissa, 2022-11-22:
My first thought is I would probably just listen to the `keydown` event for arrows and send it via serial. 

However, there are some other cases to watch out for:
1. Clicking on different locations with the mouse moves the cursor and it's a different variation of the same problem.
2. Pasting text in can present another similar problem and there are a variety of ways to do this, so it's not as easy to prevent. Here are some ways I can think of that this can be done:
    - Pressing Control-V (or Command+V on Mac or Control+Shift+V on Linux)
    - Right clicking and choosing paste
    - Going to the menu and choosing Edit -> Paste

We should be able to disable the `mousedown` event to fix the mouse cursor problem. For pasting, there's a paste event and we could either prevent anything from being inserted or just send it via serial.

This seems like an interesting challenge. The logic shouldn't add too much to the size.

tannewt, 2022-11-23:
I don't think this is important to do because it only impacts the offline serial terminal. I'd rather focus on code.cp.org

makermelissa, 2022-11-23:
Right, that's what I'm focusing on. I was just adding my thoughts to the mix.