adafruit/circuitpython #6610: Web Workflow on ESP32-S2 chips doesn't respond

RetiredWizard, 2022-07-16:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-51-ge9dd25412-dirty on 2022-07-14; FeatherS2 with ESP32S2
```

### Code/REPL

```python
No code.py
```

### Behavior

Web Workflow sites /, /fs/, /cp/serial are non-responsive after board boots and connects to wifi. A ping to the board is successful.

### Description

This has been tested on several ESP32-S2 boards with similar results. Connecting to the serial REPL via USB and pressing alternating enters, CTRL-Ds will coax the workflow forward in small steps, however multiple Ctrl-D's pressed will often cause the board to crash with a serial message of `Assertion 'num_triggered >= 0' failed, at file Socket.c:94` displayed.

If the serial console is coaxed into operation via CTRL-D/enter operations, it will continue to operate normally. Use of the serial console will also permit the other workflow pages to operate, however only while the serial terminal is actively processing input or displaying output.

tannewt, 2022-07-25:
@RetiredWizard would you mind testing with latest main? 6e7fdcc24b2aec8397528577d74a04bf36226ded fixed an mdns issue on S2 boards.

RetiredWizard, 2022-07-26:
The latest build is working much better on the Feather ESP32-S2, If the REPL is displaying the message `Press any key to enter the REPL. Use CTRL-D to reload.` the web workflow seems to work fine. :1st_place_medal: 

However, if I'm at the >>> REPL prompt or a python program is running the workflow is unresponsive.

Flash I/O no longer has any impact on moving the web workflow forward, however it seems that if the circuitpython code performs internet requests the workflow will move forward. This line of code:

     response = https.get("https://finance.yahoo.com/quote/%5EIXIC")

caused a stalled web workflow page to show the initial welcome text but none of the fields were populated.

Then during the following code loop the fields were populated and as long as the loop was running the links to the file manager and serial terminal worked.

     nasdaq = str(b''.join(response_window)).find('data-symbol="^IXIC" data-field="regularMarketChangePercent"')
     while nasdaq == -1:
         for i in range(3):
             response_window[i] = response_window[i+1]
         response_window[3] = next(iter(response.iter_content(chunk_size=256)))
         nasdaq = str(b''.join(response_window)).find('data-symbol="^IXIC" data-field="regularMarketChangePercent"')

For context, the code leading up to the loop is:

     # SPDX-FileCopyrightText: 2021 jfabernathy for Adafruit Industries
     # SPDX-License-Identifier: MIT

     # adafruit_requests usage with a CircuitPython socket
     # this has been tested with Adafruit Metro ESP32-S2 Express

     import ssl
     import wifi
     import socketpool
     from os import getenv

     import adafruit_requests as requests

     # Get wifi details and more from a .env file
     if getenv('CIRCUITPY_WIFI_SSID') is None:
         raise Exception("WiFi secrets are kept in .env, please add them there!")

     print("Connecting to %s" % getenv('CIRCUITPY_WIFI_SSID'))
     wifi.radio.connect(getenv('CIRCUITPY_WIFI_SSID'), getenv('CIRCUITPY_WIFI_PASSWORD'))
     print("Connected to %s!" % getenv('CIRCUITPY_WIFI_SSID'))
     print("My IP address is", wifi.radio.ipv4_address)

     socket = socketpool.SocketPool(wifi.radio)
     https = requests.Session(socket, ssl.create_default_context())

     TEXT_URL = "https://finance.yahoo.com/quote/%5EIXIC"
     headers = {"user-agent": "RetiredWizard@circuitpython/8.0.0A1"}

     print("Fetching text from %s" % TEXT_URL)
     response = https.get(TEXT_URL)
     print("-" * 40)
     response_window = []
     for _ in range(4):
         response_window.append(next(iter(response.iter_content(chunk_size=256))))
     print("Text Response: ", str(b''.join(response_window))[0:800])
     print("-" * 40)
     print()

RetiredWizard, 2022-07-26:
Once you get connected to the workflow serial terminal, unlike the other workflow page, it continues to operate regardless of weather you are at the REPL prompt or running python code.

I also tested a Feather ESP32-S3 board and it behaved exactly the same so this doesn't appear to be just an S2 issue.

commit 931757f4 (2022-07-27): Improve web workflow responsiveness

1. Run the socket select task at the same priority as CP. This is
  needed because it queues up the background work. Without it, CP
  needed to sleep to let the lower priority task go.
2. Close the active socket on disconnect. This prevents looping
  over a disconnected but not closed socket.

Fixes #6610. Fixes #6613
closed 2022-07-27 by tannewt