adafruit/circuitpython #6612: Installation of adafruit-circuitpython-waveshare_rp2040_zero-en_US-7.3.1.uf2 fails to connect

gitterman, 2022-07-16:
### CircuitPython version

```python
adafruit-circuitpython-waveshare_rp2040_zero-en_US-7.3.1
```

### Code/REPL

```python
Installation does not connect
```

### Behavior

Installation does not connect

### Description

I just installed adafruit-circuitpython-waveshare_rp2040_zero-en_US-7.3.1.uf2 from https://circuitpython.org/board/waveshare_rp2040_zero/ on my waveshare pico zero
and after that, thonny reports:
_Couldn't find the device automatically.
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter._
I then tried rp2-pico-20220618-v1.19.1.uf2 and this worked as it should.

dhalbert, 2022-07-17:
This is probably a Thonny problem. Thonny doesn't recognize the board as a CircuitPython board. I think it may only do that if the USB VID/PID match a list it keeps. You can click on the mode in the lower right hand corner and switch to CIrcuitPython manually. That should get you going.

gitterman, 2022-07-17:
Thanks for the hint.
This did the trick.

closed 2022-07-17 by gitterman