adafruit/circuitpython #6613: Web Workflow on ESP32 V2 feather chips doesn't respond in specific scenario

jepler, 2022-07-16:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-54-g8b30f0120-dirty on 2022-07-14; Adafruit Feather ESP32 V2 with ESP32
```

### Code/REPL

```python
# None
```

### Behavior

When you follow a specific set of steps, the circuitpython.local site becomes unresponsive

### Description

 1. load the /cp/serial page in the browser
 2. Click the Ctrl-C button in the web page, so that it is sitting at the ">>>" REPL
 3. Use the browser's reload button

This hangs waiting to load /cp/serial/. The only thing that seems to make it continue is pressing ctrl-d in a terminal connected to the USB serial port.

tannewt, 2022-07-25:
What browser?

jepler, 2022-07-26:
firefox on linux

RetiredWizard, 2022-07-26:
This same behavior is happening on S2 and S3 chips. I suspect this is a subset of the issues I'm seeing in #6610

commit 931757f4 (2022-07-27): Improve web workflow responsiveness

1. Run the socket select task at the same priority as CP. This is
  needed because it queues up the background work. Without it, CP
  needed to sleep to let the lower priority task go.
2. Close the active socket on disconnect. This prevents looping
  over a disconnected but not closed socket.

Fixes #6610. Fixes #6613
closed 2022-07-27 by tannewt