adafruit/circuitpython #6617: Changing from Wippersnapper firmware to CircuitPython drive name remains WIPPER

prcutler, 2022-07-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1 and Adafruit CircuitPython 7.3.1 on an a QTPY-ESP32-S2
WipperSnapper 1.0 beta 38 firmware
```

### Code/REPL

```python
N/A
```

### Behavior

I have a QTPY-ESP32-S2 running Wippersnapper 1.0 beta 38 firmware.  I want to reinstall CircuitPython:

After putting the QTPY in boot mode and installing either CIrcuitPython 7.3.1 or 8.0-alpha, the drive name remains "WIPPER" and does not change back to "CIRCUITPY" as expected.

### Additional information

I am able to manually rename the drive in macOS and the drive name remains even after changing the firmware from 7.3.1 to 8.0-alpha

tannewt, 2022-07-25:
@brentru Is wipper using FATFS? CircuitPython must be reusing the partition.

@prcutler It is intended to not be CIRCUITPY if the existing file system has a different name. We don't want to mess with the FS if it exists already.

brentru, 2022-07-27:
@tannewt WipperSnapper uses FatFS for filesystem-backed provisioning.

tannewt, 2022-07-27:
Ok, this is as intended then. @prcutler let me know if you think we should do something else.

closed 2022-07-27 by tannewt

prcutler, 2022-07-27:
By reinstalling a CircuitPython UF2, I would expect the drive to now be called "CIRCUITPY" - doesn't things like Mu rely on that?

tannewt, 2022-07-28:
They shouldn't rely on the CIRCUITPY name because users can change it.