adafruit/circuitpython #6621: Provide means for reliably relating serial port and filesystem mount path

aivarannamaa, 2022-07-20:
It would be nice if one (a person or a tool) could reliably locate device's filesystem mount path when given a serial port and vice versa.

Example situations:

* Thonny users need to select their device by serial port but as serial REPL can't be used for saving files by default, Thonny needs to locate device's filesystem mount. It currently uses `storage.getmount('/')` to retrieve volume label and hopes to find single volume with given name. This doesn't work out when there are several CP boards plugged in.
* [Pipkin](https://pypi.org/project/pipkin/) allows users to select their device by mount path. I'd like pipkin to consult devices list of frozen modules (`help("modules")`) before performing an install, so that it can see whether `adafruit_bus_device` is built-in for this device or not (here's a related discussion: https://github.com/adafruit/circuitpython-build-tools/pull/90#issuecomment-1189849977). For this it is necessary to find the serial port corresponding to given mount path.

I assume it is not possible for CircuitPython to directly report its mount path or serial port, but would it be possible to append another line to boot_out.txt, which reports device's USB serial number? If I'm not mistaken, this is a unique identifier, which means tools could use it for reliably finding the right port given the mount path and vice versa.

Neradoc, 2022-07-20:
How about [discotool](https://github.com/Neradoc/discotool) ?

dhalbert, 2022-07-20:
Or https://github.com/adafruit/Adafruit_Board_Toolkit, which uses most of the same insides as discotool?

dhalbert, 2022-07-20:
@Neradoc thanks for the clarification.

@aivarannamaa: `microcontroller.cpu.uid` returns the serial number, so you could access it from the REPL via the Thonny back end. But maybe that doesn't help to distinguish it via volume?

We have had some issues about the nibble and byte order of `cpu.uid`, and we may need to flip some things around to make them match.

aivarannamaa, 2022-07-20:
@Neradoc, thanks for the library and explanations! Somehow it never occurred to me that I could turn to OS for getting the match between serial port and mount point. Your library's deps are rather heavy, though... I'll need to give it some thought.

@dhalbert, I tried your advice on Cytron Maker Pi RP2040, but `microcontroller.cpu.uid` (`bytearray(b'\xe6'\xc0b\x13B\x155')`)  seems to be very different from its USB serial id ( `"E660C06213421535"`).

dhalbert, 2022-07-20:
> @dhalbert, I tried your advice on Cytron Maker Pi RP2040, but `microcontroller.cpu.uid` (`bytearray(b'\xe6'\xc0b\x13B\x155')`) seems to be very different from its USB serial id ( `"E660C06213421535"`).

The `bytearray` is mixed printing characters and hex-escaped characters. I think you substituted a single quote for a backtick in there. Assuming that, I get:
```py
>>> [hex(b) for b in b'\xe6`\xc0b\x13B\x155']
['0xe6', '0x60', '0xc0', '0x62', '0x13', '0x42', '0x15', '0x35']
```
whose digits match the concatenated digits in `"E660C06213421535"`.

aivarannamaa, 2022-07-20:
> It looks like with this I can really implement the mount point lookup in Thonny.

Oops, I just realized I got my thinking messed up. By learning the USB serial ID via MicroPython code I can easily find only the right serial port, which I already know. If I understood @Neradoc correctly, then the hard part is finding the mount point by serial port information. Having the serial ID explicitly written out at the mount point should make it easy.

tannewt, 2022-07-25:
I'd be ok with adding serial to boot_out.txt. Maybe it'd be better to write a json file with info we get from boot_out.txt. That'd make version agnostic parsing much easier. (It'd be bad for tiny filesystems though.)

aivarannamaa, 2022-07-26:
A line in boot_out.txt would be good enough for my projects. If I simply had to find a line with specific prefix (eg. `Serial: `), then parsing wouldn't be too much harder compared to json format.

tannewt, 2022-08-03:
I'm adding a `UID:` like with the uid printed as `%02X` per byte.

commit eb7ddf52 (2022-08-05): Fix BLE workflow and add boot_out.txt UID

Fixes #6621
closed 2022-08-05 by tannewt