adafruit/circuitpython #6622: Long startup time on ESP32-S2

deshipu, 2022-07-20:
### CircuitPython version

```python
7.3.1
```

### Code/REPL

```python
import board
import digitalio

led = digitalio.DigitalInOut(board.LED)
led.switch_to_output(value=1)

while True:
    pass
```

### Behavior

The LED comes on after 2.45s from connecting power or pressing RST button.

### Description

This seems about twice as long as expected, with the safe mode delay being about 1s, and without an UF2 bootloader on the board.

dhalbert, 2022-07-20:
Per @deshipu, this test was done with a UM FeatherS2 Neo.

tannewt, 2022-07-25:
I suspect this may be due to large RAM sizes. I'd be curious to see the timing output in the ESP-IDF debug output.

dhalbert, 2022-07-25:
There is an `sdkconfig` setting `CONFIG_SPIRAM_MEMTEST=y` which does an SPI RAM test on startup. Is this what you are thinking of?

DJDevon3, 2023-04-28:
@deshipu can revisit and retest this one on UM FeatherS2 on 8.0.5 stable (most recent).  I'm not seeing this issue anymore on my UM FeatherS2.

dhalbert, 2023-10-28:
Closing as apparently fixed.

closed 2023-10-28 by dhalbert