adafruit/circuitpython #6644: Support Webdav in the web workflow

Neradoc, 2022-07-25:
It would be nice to support some native clients in the web workflow.
Since we already have a web server, we could add (some of) the Webdav extensions to it.
That would allow accessing the files with a file manager application like Cyberduck, even mount as a drive on the computer (and run circup, why not). This in turn allows editing the code with our editor/IDE of choice, some of them might even directly support webdav.

It might not be necessary to support all of the commands, the [nginx webdav module](http://nginx.org/en/docs/http/ngx_http_dav_module.html) seems to only support PUT, DELETE, MKCOL, COPY, and MOVE for example.
The question is, what clients are there, and what do they need to be supported ?
And how much space will that take ?

Here is a list from wikipedia:
- CloudMounter from Eltima Software
- Git supports writing to HTTP remotes
- Linux via GVfs, including GNOME Files and via KIO, including Konqueror and Dolphin
- macOS, including native support for CalDAV and CardDAV, the design of which is based on WebDAV
- Microsoft Windows and Microsoft Office, including native support in Explorer, via the WebDAV Redirector[22]
- NetDrive from Bdrive, Inc.
- WebDrive from South River Technologies
- WinSCP from Martin Přikryl

IrregularShed, 2023-11-07:
I dreamt about this last night (exciting, huh?). It would a huge quality of life improvement for non-native USB devices. This led me to ponder FUSE on Linux/Mac and WinFsp on Windows, but the much more native WebDAV would be preferable for portability between systems.

vvuk, 2024-09-22:
From a quick glance -- this should be _fairly_ straightforward to implement, given that the web workflow is all implemented in straight C (i.e. there is no "web server" framework, it's all directly implemented). The only complication is probably `PROPFIND` -- DAV PROPFIND works with XML documents, including proper namespace definitions and such, meaning that at least a very lightweight XML namespace aware xml parser would need to be present. Something like [`yxml`](https://dev.yorhel.nl/yxml) might be a good fit for that.