adafruit/circuitpython #6653: Web Workflow Serial Terminal screen stops functioning

RetiredWizard, 2022-07-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-69-gbc926b041-dirty on 2022-07-26; Adafruit QT Py ESP32C3 with ESP32-C3FN4
Adafruit CircuitPython 8.0.0-alpha.1-69-gbc926b041-dirty on 2022-07-26; Adafruit Feather ESP32S2 with ESP32S2
Adafruit CircuitPython 8.0.0-alpha.1-69-gbc926b041-dirty on 2022-07-26; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3

Circuit Python Built from Pull Request #6651
```

### Code/REPL

```python
cmd = ""
while cmd.upper() != "EXIT":
    cmd = input(":>")
```

### Behavior

When each of the boards boots you are able to connect to them over the Web Workflow and open a serial terminal window. On boot the code.py program is running and the serial terminal window can be refreshed without problem.

However refreshing the serial terminal screen at certain points will disconnect the terminal (the C, D and SEND buttons go grey and input to the command line is blocked). The following message is displayed on the browser Dev Tool console:

[image]

Connecting to REPL via the USB serial port of a board in this state appears to show some of the header bar displayed in the terminal output. If the Workflow serial terminal attempted to display the header string/icons as terminal text that might account for the UTF-8 decode error.

[image]

Once this error occurs new serial terminal windows will not connect and the existing window continues to error if refreshed. A board hard reset or power cycle appears to be the only way to return the serial terminal back to operation.

### Description

The most consistent way I've been able to reproduce the error is to type exit into the command line and then press the `SEND` button (don't use the enter key). Once the Press any key to enter the REPL message is displayed, refresh the browser screen. Usually refreshing once will cause the error which sometimes takes a minute to occur, but if the refresh works once, pressing refresh a second time will usually trigger the error.

The S3 board seems to throw the error the quickest and in more situations than the others.

### Additional information

I did this testing using builds from PR #6651

RetiredWizard, 2022-07-27:
I have seen the "Could not decode a text frame as UTF-8" dev console message a number of times when the serial console stops working but it's difficult to nail down reproducible steps. I think the following REPL commands do cause the error on all three chip types:

     a = b'*'*160
     print(str(a))

tannewt, 2022-07-28:
There is 125 byte boundary for the websocket frame length. I may be encoding the higher number wrong. Will look now. Thanks!

PR #6666 (2022-07-29): Lots of web workflow, C3 and title bar fixes
* Fixes #6221 - C3 hang on `import wifi`. Enabling the WiFi PHY was
  disabling USB. Now boards that use it set CONFIG_ESP_PHY_ENABLE_USB
  explicitly.
* Fixes #6655 - Allows pasting into the web serial page. Fixes reading
  more than 0xf bytes at a time.
* Fixes #6653 - Fixes web socket encoding of payloads >125 bytes. Can
  happen when printing a long string.
* Fixes C3 responsiveness when waiting for key to enter REPL. (It
  now correctly stops sleeping.)
* Disables title bar updates when in raw REPL. Related to #6548.
* Adds version to title bar [... cut]
commit d6344812 (2022-07-29): Lots of web workflow, C3 and title bar fixes

* Fixes #6221 - C3 hang on `import wifi`. Enabling the WiFi PHY was
  disabling USB. Now boards that use it set CONFIG_ESP_PHY_ENABLE_USB
  explicitly.
* Fixes #6655 - Allows pasting into the web serial page. Fixes reading
  more than 0xf bytes at a time.
* Fixes #6653 - Fixes web socket encoding of payloads >125 bytes. Can
  happen when printing a long string.
* Fixes C3 responsiveness when waiting for key to enter REPL. (It
  now correctly stops sleeping.)
* Disables title bar updates when in raw REPL. Related to #6548.
* Adds version to title ba [... cut]
closed 2022-07-29 by dhalbert