adafruit/circuitpython #6654: Web Workflow file manager screen occasionally doesn't list files

RetiredWizard, 2022-07-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-69-gbc926b041-dirty on 2022-07-26; Adafruit QT Py ESP32C3 with ESP32-C3FN4
Adafruit CircuitPython 8.0.0-alpha.1-69-gbc926b041-dirty on 2022-07-26; Adafruit Feather ESP32S2 with ESP32S2
Adafruit CircuitPython 8.0.0-alpha.1-69-gbc926b041-dirty on 2022-07-26; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3

Circuit Python Built from Pull Request #6651
```

### Code/REPL

```python
cmd = ""
while cmd.upper() != "EXIT":
    cmd = input(":>")
```

### Behavior

Upon opening or refreshing the Web Workflow file manager, occasionally the displayed directory will be blank and no files will be listed.

[image]
Usually the browser dev tools console displays the following message:

[image]

The following messages have also been displayed though:

[image]
[image]

### Description

This issue is easiest to reproduce on the ESP32-C3 but I have seen it happen on both the S2 and S3. In my testing of the C3 board the networking has been getting better, but has been somewhat hit or miss. I may be at the edge of it's range from my wireless router.

### Additional information

I did this testing using builds from PR https://github.com/adafruit/circuitpython/pull/6651

tannewt, 2022-08-04:
@RetiredWizard Are you still seeing this?

RetiredWizard, 2022-08-04:
I am, but I"m not sure I'm running the latest bits, let me grab them for my S2 and C3 and give it another try

RetiredWizard, 2022-08-04:
It's still happening on both boards using the newest builds from the S3 site. It takes a lot of refreshing to get it to happen on the S2 but it happens pretty quickly on the C3. The C3 takes a bit to get it connected to the WiFi, I typically have to place it closer to the router and give it a couple power.

Given the less robust networking going on with the C3, I'm thinking that perhaps a network error is triggering the error.

RetiredWizard, 2022-08-04:
I think @jerryn suggested that a weak WiFi signal creates some intermittent issues, and that's probably what's going on here.

I was thinking that perhaps the `net::ERR_INVALID_CHUNKED_ENCODING 200 (OK)` error could be caught and either a message displayed in the browser window or a partial directory listing displayed but perhaps just understanding that a poor WiFi connection = intermittent issues is all there is to be done at this point.

tannewt, 2022-08-05:
Could you try it with a curl command? I think the examples have detailed logging on. It may make it clearer what the issue is.

RetiredWizard, 2022-08-06:
C:\>curl -v -u :CPPasswd -H "Accept: application/json" -L --location-trusted http://10.0.0.206/fs/
     *   Trying 10.0.0.206:80...
     * Connected to 10.0.0.206 (10.0.0.206) port 80 (#0)
     * Server auth using Basic with user ''
     > GET /fs/ HTTP/1.1
     > Host: 10.0.0.206
     > Authorization: Basic OkNQMTIz
     > User-Agent: curl/7.84.0
     > Accept: application/json
     >
     * Mark bundle as not supporting multiuse
     < HTTP/1.1 200 OK
     < Transfer-Encoding: chunked
     < Content-Type: application/json
     < Access-Control-Allow-Credentials: true
     < Vary: Origin, Accept, Upgrade
     < Access-Control-Allow-Origin:
     <
     [{"name": "MyDOS.py","directory": false, "modified_ns": 946692482000000000, "file_size": 4536 },{"name":  "boot_out.txt","directory": false, "modified_
     ns": 946684806000000000, "file_size": 207 },{"name": "writefile.py","directory": false, "modified_ns": 946685532000000000, "file_size": 256 },{"name":
      "basicpython.py","directory": false, "modified_ns": 946685554000000000, "file_size": 2995 },{"name": "lib","directory": true, "modified_ns": 94668480
     4000000000, "file_size": 0 },{"name": "blink.py","directory": false, "modified_ns": 946685564000000000, "file_size": 3376 },{"name": "boot.py","direct
     ory": false, "modified_ns": 946685574000000000, "file_size": 3757 },{"name": "cls.py","directory": false, "modified_ns": 946685574000000000, "file_siz
     e": 22 },{"name": "main.sav","directory": false, "modified_ns": 946686148000000000, "file_size": 14 },{"name": "diff.py","directory": false, "modified
     _ns": 946685598000000000, "file_size": 6655 },{"name": "edit.py","directory": false, "modified_ns": 946685598000000000, "file_size": 157 },{"name": "f
     ileview.py","directory": false, "modified_ns": 946685652000000000, "file_size": 4951 },{"name": "fsedit.py","directory": false, "modified_ns": 9466851
     68000000000, "file_size": 157 },{"name": "i2cscan.py","directory": false, "modified_ns": 946685654000000000, "file_size": 935 },{"name": "lcdprint.py"
     ,"directory": false, "modified_ns": 946685662000000000, "file_size": 2220 },{"name": "lcdscroll.py","directory": false, "modified_ns": 946685674000000
     000, "file_size": 4217 },{"name": "code.py","directory": false, "modified_ns": 946685574000000000, "file_size": 72 },{"name": "piano.py","directory":
     false, "modified_ns": 946685688000000000, "file_size": 4708 },{"name": "prtglbvar.py","directory": false, "
     9
     946685690000000000, "file_size": 404 },{"PyDOS.py","directory":
     5
     fal* Malformed encoding found in chunked-encoding
     * Closing connection 0
     curl: (56) Malformed encoding found in chunked-encoding

RetiredWizard, 2022-08-06:
A little more detail from the dev console screen as well:
[image]

RetiredWizard, 2022-08-06:
A slightly different message from curl:

     modified_ns": 946686086000000000, "file_size": 4750 },{"name": "menu.txt","directory": false, "modified_ns": 
     946686098000000000, "file_size": 101 },{"
     name": "cp","directory":
     5
     false
     2* Illegal or missing hexadecimal sequence in chunked-encoding
     * Closing connection 0
     curl: (56) Illegal or missing hexadecimal sequence in chunked-encoding

RetiredWizard, 2022-08-06:
These error are really pretty easy to generate on the C3, it takes a lot more work on the other boards. I have the C3 powered from a phone charger right next to the WiFi router.

tannewt, 2022-08-09:
I'm not able to reproduce this on a C3. Is there a code.py running at the same time? Is there another web request occurring?

RetiredWizard, 2022-08-09:
I typically do have a code.py application running but it's just sitting at an input statement. I bring up the welcome screen and then the file manager. If the file manager loads correctly, it usually only takes a few page refreshes for the error to occur on the C3. I'll download the latest bits from S3 and see if anything has changed.

RetiredWizard, 2022-08-09:
I loaded the latest S3 bits, killed my code.py application so we're sitting at the REPL prompt and then started refreshing the file manager. After six refreshes I hit an error but the console message was not one of the usual ones:

[image]

I then started hitting refresh again and after another 10 or so refreshes I got another miss-load with the typical `ERR_INVALID_CHUNKED_ENCODING 200 (OK)` developer console message.

One thing that occurs to me is that I have 77 files/directories in the / directory so the file manager is displaying lots of data with each refresh. That might be the difference.

tannewt, 2022-08-09:
Ok, I added some to mine but not 77. It'd also be helpful to see the DEBUG=1 output of CP when this happens.

RetiredWizard, 2022-08-09:
I can compile a DEBUG=1 image, Will the debug output show up on the serial REPL or do I need to enable debug uart pins as well?

tannewt, 2022-08-09:
On C3 it is the separate UART pins.

tannewt, 2022-08-09:
I'm going to unblock 8 on this. It'd be good for others to chime in if they see this.

RetiredWizard, 2022-08-10:
In order to attach the uart pins to my PC I had to move the C3 away from the router. It took a dozen or so power cycles to get the C3 to connect to Web Workflow over Wifi and when it did almost every refresh on the file manager had the error. If I refreshed enough times, I could eventually get a successful file manager display.

The debug uart output from one of the failed refreshes is as follows. The `short send` messages actually filled several screens but I only included the first  dozen or so.

     I (286480) CP webserver: Request GET /fs/
     I (286490) CP webserver: Header Host 10.0.0.206
     I (286500) CP webserver: Header Connection keep-alive
     I (286520) CP webserver: Header Authorization Basic OkNQMTIz
     I (286530) CP webserver: Header Accept application/json
     I (286580) CP webserver: Header User-Agent Mozilla/5.0 (Windows NT 6.3; Win64; x64) AppleWebKit/537.36 (KH
     I (286580) CP webserver: Header DNT 1
     I (286590) CP webserver: Header Referer http://10.0.0.206/fs/
     I (286610) CP webserver: Header Accept-Encoding gzip, deflate
     I (286620) CP webserver: Header Accept-Language en-US,en;q=0.9
     E (286720) CP webserver: short send 7 11
     E (286780) CP webserver: short send -104 1
     E (286780) CP webserver: short send -11 2
     E (286780) CP webserver: short send -11 9
     E (286780) CP webserver: short send -11 2
     E (286780) CP webserver: short send -11 1
     E (286790) CP webserver: short send -11 2
     E (286790) CP webserver: short send -11 11
     E (286800) CP webserver: short send -11 2
     E (286800) CP webserver: short send -11 1
     E (286810) CP webserver: short send -11 2
     E (286810) CP webserver: short send -11 13
     E (286820) CP webserver: short send -11 2
     E (286820) CP webserver: short send -11 1
     E (286830) CP webserver: short send -11 2

RetiredWizard, 2022-08-10:
This is an entire debug output from a failed file manager refresh.

[DEBUG_output.txt](https://github.com/adafruit/circuitpython/files/9295584/DEBUG_output.txt)

makermelissa, 2022-08-10:
I've occasionally noticed no files showing as well. Interestingly enough, they were showing just fine on the code.circuitpython.org changes I was making (from a json request call) despite not showing on the local web server.

tannewt, 2022-08-10:
Ok, I have a theory about this. We're filling up the outgoing socket and therefore we're only partially sending data. That's this line:

>  E (286720) CP webserver: short send 7 11

When two of these occur, we're dropping some characters in between and are no longer correct in the response. As a result, the browser stops us.

PR #6742 (2022-08-13): Retry send if only some bytes sent
Fixes #6654 and fixes #6689
commit 0b098f50 (2022-08-13): Retry send if only some bytes sent

Fixes #6654 and fixes #6689
closed 2022-08-13 by MicroDev1