adafruit/circuitpython #6671: Web Workflow file browser edit function doesn't work in folders

RetiredWizard, 2022-07-30:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-85-g462eb7157 on 2022-07-29; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
None
```

### Behavior

In the Web Workflow file browser, if you navigate to a folder and attempt to edit a file a `loading status: 404` message on top of an empty edit window is displayed. The browser Developer tools console window displays:

[image]

tannewt, 2022-08-01:
Are we missing the folder name from the edit url? What is the edit url for you?

RetiredWizard, 2022-08-01:
Here is a screenshot of the URL after clicking on the Edit button:
[image]

FoamyGuy, 2022-08-01:
Yes, I think the issue is missing the folders in the edit links.

For the file `/lib/adafruit_logging.py` We have this URL:
```
http://192.168.1.113/edit/#adafruit_logging.py
```
Which is missing `/lib/` in the hash portion of the URL.

The edit page itself does seem to support the sub directory if you construct the URL yourself though. This URL does allow me to edit the file inside of it's directory successfully:
```
http://192.168.1.113/edit/#/lib/adafruit_logging.py
```

I'll look into the links on the directory page, that is where it would need to get the full path inserted into the hash rather than just the filename.

PR #6680 (2022-08-02): fix edit URL for files inside of directories
Fixes the URL for the edit link on directory pages in the web workflow to include the directory in the hash portion of the URL.

Resolves: #6671 

Under current version the file `/lib/adafruit_logging.py` will end up with a url containing only `#adafruit_logging.py`

With this branch it will include the directory so it will be `#/lib/adafruit_logging.py`
closed 2022-08-02 by tannewt