adafruit/circuitpython #6673: cpy rtc module function datetime returns other day-of-the-week value than received from io.adafruit time service

PaulskPt, 2022-07-30:
### CircuitPython version

```python
Various versions but now using: Adafruit CircuitPython 8.0.0-alpha.1
```

### Code/REPL

```python
import rtc
r = rtc.RTC()
# [...]
# Below the commands to request and receive response to/from io.adafruit.com 
TIME_URL = "https://io.adafruit.com/api/v2/{:s}/integrations/".format(aio_username)
TIME_URL += "time/strftime?x-aio-key={:s}".format(aio_key)
TIME_URL += "&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z"
# [...]
response = requests.get(TIME_URL)
resp_lst = response.text.split(" ")

# REPL output: response.text.split= ['2022-07-30', '19:05:34.452', '211', '6', '+0100', 'WEST']

dt = response.text[:10]
tm = response.text[11:16] # 23]
hh = int(resp_lst[1][:2])
mm = int(resp_lst[1][3:5]) # +mm_corr # add the correction
ss = int(resp_lst[1][6:8])
yd = int(resp_lst[2]) # day of the year
wd = int(resp_lst[3])
if resp_lst[5] == 'WEST':
    dst = 1
elif resp_lst[5] == 'WET':
    dst = 0
else:
    dst = -1

# [...]
# Below commands to set the built-in RTC of the MAGTAG
tm2 = (yy, mo, dd, hh, mm, ss, wd, yd, dst)
tm3 = time.struct_time(tm2)
r.datetime = tm3  # set the built-in RTC

# Retrieve the current datetime from the built-in RTC
ct = r.datetime
# weekday
wd = ct[tm_wday]
print("RTC current_time=", ct)
# REPL result
RTC current_time= struct_time(tm_year=2022, tm_mon=7, tm_mday=30, tm_hour=19, tm_min=14, tm_sec=0, tm_wday=5, tm_yday=211, tm_isdst=-1)
```

### Behavior

The io.adafruit.com time service response for Saturday 2022-07-30 19:14:00  (timezone utc+1) gives for 
'day-of-the-week' a value of 6 (because it's base is Monday = 1.
['2022-07-30', '19:05:34.452', '211', '6', '+0100', 'WEST']

The CPY r.datetime() returns a time.struct_time in which, for the same day, a value of 5 for the 'day-of-the-week' because it's 
base is Monday = 0

I have to 'fix' this difference by subtracting '1' from the 'day-of-the-week' value received from the aio time service:
wd = int(resp_lst[3])-1

See annotated screenshot of REPL output:

[image]

Beside this I noticed that r.datetime() is_dst always returns '-1' whatever value was entered while setting the RTC.

### Description

The bug does not cause crashes or errors.
It only causes difference because io.adafruit.com time service uses a different base value for 'date-of-the-week' than does the Circuitpython rtc.datetime() function.

### Additional information

MAGTAG flashed with TinyUF2 Bootloader 0.8.0 - tinyusb (0.12.0-203-ga4cfd1c6)
Model: Adafruit MagTag 2.9 Grayscale
Board-ID: ESP32S2-MagTag_29gray-revC
Date: Jan  5 2022
MAGTAG flashed with Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-09; Adafruit MagTag with ESP32S2

Perhaps this is not a bug, however I think that both io.adafruit.com time service as the Circuitpython rtc module, function datetime, should result in a same value for the 'day-of-the-week' item for the same calender date.

PaulskPt, 2022-07-31:
I just found this article on adafruit learn: ```rtc-with-circuitpython```. In the example the tm_wday is based on Sunday = 0. 
[image]

Neradoc, 2022-07-31:
Yep, sunday is 0 in struct_time in python:
https://docs.python.org/3/library/time.html#time.struct_time
Other implementations might use a different way of counting.

The weekday is computed by the time module, it doesn't matter what value you give it.
Note that when you set the RTC you can then use the time module to get the time.
```py
>>> rtc.RTC().datetime = time.struct_time((2022,7,30,19,14,0,5,211,-1))
>>> time.localtime()
struct_time(tm_year=2022, tm_mon=7, tm_mday=30, tm_hour=19, tm_min=14, tm_sec=1, tm_wday=5, tm_yday=211, tm_isdst=-1)
>>> rtc.RTC().datetime = time.struct_time((2022,7,30,19,14,0,6,211,-1))
>>> time.localtime()
struct_time(tm_year=2022, tm_mon=7, tm_mday=30, tm_hour=19, tm_min=14, tm_sec=1, tm_wday=5, tm_yday=211, tm_isdst=-1)
>>> rtc.RTC().datetime = time.struct_time((2022,7,30,19,14,0,1_000_000,211,-1))
>>> time.localtime()
struct_time(tm_year=2022, tm_mon=7, tm_mday=30, tm_hour=19, tm_min=14, tm_sec=1, tm_wday=5, tm_yday=211, tm_isdst=-1)
```

DST can not be computed, so it is ignored and always -1.
Hmmm I thought that was mentioned in the docs, but apparently not.
https://github.com/adafruit/circuitpython/blob/462eb71570819a315942f808c8d6839b5904662c/shared-bindings/time/__init__.c#L163-L173

PaulskPt, 2022-08-01:
@Neradoc thank you for your explanations. Yes, I know how to use the time module. I have done different projects using external and internal RTC on differen microcontrollers. I just 'stumbled' over the difference of 'day-of-the-week' value as explained above. Guess have to live with that and - as written above - add a correction of -1 on the wday received from the aio time service.
For me case closed

closed 2022-08-01 by PaulskPt

Neradoc, 2022-08-01:
> Guess have to live with that and - as written above - add a correction of -1 on the wday received from the aio time service.

You set the time from the aio time service, and then get the time with time.localtime().
At no time do you need to care what the wday of aio is.