adafruit/circuitpython #6675: Data in displayio.Bitmap is mirrored for depths < 8

deshipu, 2022-08-01:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.0-dirty on 2022-07-26; S2Mini with ESP32S2-S2FN4R2
```

### Code/REPL

```python
>>> b = displayio.Bitmap(8, 8, 65526)
>>> m = memoryview(b)
>>> b[0, 0] = 1
>>> m[0]
1
>>> b = displayio.Bitmap(8, 8, 256)
>>> m = memoryview(b)
>>> b[0, 0] = 1
>>> m[0]
1
>>> b = displayio.Bitmap(8, 8, 16)
>>> m = memoryview(b)
>>> b[0, 0] = 1
>>> m[0]
0
>>> m[1]
0
>>> m[2]
0
>>> m[3]
16
```

### Behavior

When the depth is less than 8, the data inside the bitmap is saved right to left instead of left to right. This makes writing image loaders harder, as we have to consider every depth separately, instead of just dumping the image data into the bitmap.

tannewt, 2022-08-01:
I don't think this is intentional. My brain hurt after writing all that code. I'd be happy to merge in a change for this. I agree it should be consistent.

deshipu, 2022-08-01:
Bits are painful, I will see if I can get it consistent without making the code bigger.