adafruit/circuitpython #6676: alarm.time.TimeAlarm does not work when sleep is called in conjunction with alarm.pin.PinAlarm

pookzilla, 2022-08-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.2 on 2022-07-20; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
import board
import alarm
import time
from digitalio import DigitalInOut, Direction, Pull

led = DigitalInOut(board.LED)
led.direction = Direction.OUTPUT
led.value = False
time.sleep(10)
print("Waking up")
time_alarm = alarm.time.TimeAlarm(monotonic_time = time.monotonic() + 20)
pin_alarm = alarm.pin.PinAlarm(pin=board.D12, value=True, pull=True)
print("Going to sleep")
alarm.exit_and_deep_sleep_until_alarms(time_alarm, pin_alarm)
# alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

### Behavior

If you call either of the alarm.sleep functions with both a pin alarm and a time alarm then only the pin alarm is functional - the time alarm will never be executed.  

### Description

If you are connected to serial monitor while executing this you can also note that the usual "Pretending to deep sleep until alarm" message does not happen when both alarms are used.

kattni, 2022-09-21:
I did not run into this when I did the same thing a couple of weeks ago. I will test with this code to verify - it's possible we resolved the issue in 8.x.

kattni, 2022-10-04:
`TimeAlarm` and `PinAlarm` work fine together if the code is updated so the `value=True` parameter for `PinAlarm` is changed to `value=False` and you connect pin D12 to GND instead of 3V. The issue is with the way `PinAlarm` works, not with using the two types of pins together. This issue is resolved.

I have created a new issue for the `PinAlarm`-specific issue: #6994.

closed 2022-10-04 by kattni