adafruit/circuitpython #6679: ESP32-S3 builds don't boot into Circuit Python

RetiredWizard, 2022-08-02:
### CircuitPython version

```python
adafruit-circuitpython-adafruit_feather_esp32s3_4mbflash_2mbpsram-en_US-20220801-664cadc.uf2
Absolute newest from S3
```

### Code/REPL

```python
N/A
```

### Behavior

No REPL, No CIRCUITPY USB drive, no Serial Port.

Lights on the Feather ESP32-S3 when powered up: 1 Purple Flash, 4 rapid Yellow flashes, Green flash every 5 seconds.

### Additional information

Also built a DEBUG-1 image for the ESP32-S3-DevkitC-1-N8R2 with same results. On power up, 4 rapid Yellow flashes, then nothing.

On the UART port the following is displayed upon power on but nothing else:

     ESP-ROM:esp32s3-20210327
     Build:Mar 27 2021
     rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
     SPIWP:0xee
     mode:DIO, clock div:2
     load:0x3fcd0108,len:0x132c
     load:0x403b6000,len:0xb80
     load:0x403ba000,len:0x2ec0
     entry 0x403b6238

kmatch98, 2022-08-02:
Not sure if it’s relevant to your issue, but on the ESP32-S3 devkit N8R8 boards, after I flash the .bin file to the UART port using esptool, I have to plug into the “USB” port to access the REPL and CIRCUITPY drive.

RetiredWizard, 2022-08-02:
@kmatch98  are you using the "Absolute newest" bits from S3? I don't even get a serial port device listed from the USB port.

kmatch98, 2022-08-02:
@RetiredWizard To confirm, I just took a fresh ESP32-S3 devkit N8R8 board, connected the UART port and loaded the UF2 bootloader [as described in these instructions at the bottom right of the page](https://circuitpython.org/board/espressif_esp32s3_devkitc_1_n8r8/).

I switched to the "USB" port and the `S3DKC1BOOT` drive appeared.

Then I copied the latest UF2 file `adafruit-circuitpython-espressif_esp32s3_devkitc_1_n8r8-en_US-20220801-664cadc.uf2`.  Results were as-expected, with a `CIRCUITPY` drive and saw the REPL on the serial terminal.  

Let me know if I can help anyhow.

RetiredWizard, 2022-08-02:
Okay, I was using esptool to flash the .bin files and I probably had one of the parameters wrong again :/ When I originally read the boot loader install instructions I took the note about update.uf2 not available on the S3 to mean that I couldn't use the UF2 boot loader. 

After installing the UF2 boot loader and grabbing the latest UF2 file the devkit is running fine. I'll take another look at my feathers3, but this obviously isn't a generic S3 issue and the Feather ESP32-S3 problem I was having was probably user error as well.

I'm going to go ahead and close this. Thanks @kmatch98 !!!

kmatch98, 2022-08-02:
For posterity, here is the `esptool` command that I use to flash my ESP32-S3 boards:

```
python3 -m esptool -b 921600 --chip esp32s3 write_flash 0x0 build-espressif_esp32s3_devkitc_1_n8r8/firmware.bin
```

RetiredWizard, 2022-08-02:
That's what I was using as well, IDK....

RetiredWizard, 2022-08-02:
@kmatch98 On the S3 Devkit you have running with the latest bits, can you jump onto the REPL and `import wifi` ?

Neradoc, 2022-08-02:
Mine crashed, didn't come back on its own.
```
$ tio /dev/cu.usbmodemC7FD1A0EA9CF1
[19:36:35.209] tio v1.47
[19:36:35.210] Press ctrl-a q to quit
[19:36:35.213] Connected
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.0.0-alpha.1-86-g664cadc3a on 2022-08-01; ESP32-S3-DevKitC-1-N8R8 with ESP32S3
>>> import wifi

[19:36:42.439] Disconnected
[19:36:43.442] Warning: Could not open tty device (No such file or directory)
[19:36:43.442] Waiting for tty device..
```
Open another issue with that ?

PR #6687 (2022-08-03): Fix S2/S3 USB
The new Kconfig that fixed C3 added a dependency on USB_OTG_SUPPORTED
that is only calculated if we include the usb module in cmake.

Fixes #6679
commit 1c69fe17 (2022-08-03): Fix S2/S3 USB

The new Kconfig that fixed C3 added a dependency on USB_OTG_SUPPORTED
that is only calculated if we include the usb module in cmake.

Fixes #6679
closed 2022-08-03 by gamblor21