adafruit/circuitpython #6689: Web Workflow serial terminal occasionally disconnects

RetiredWizard, 2022-08-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-89-gb38109c48-dirty on 2022-08-02; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
Adafruit CircuitPython 8.0.0-alpha.1-89-gb38109c48-dirty on 2022-08-02; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
Adafruit CircuitPython 8.0.0-alpha.1-89-gb38109c48-dirty on 2022-08-02; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
from supervisor import runtime
print("Press any key to stop output")
while runtime.serial_bytes_available == 0:
    print("0123456789"*5+"abcdefghijklmnopqrstuvwxyz")
```

### Behavior

Usually Within a 10 or 20 seconds or so of starting this python script the Web Workflow serial terminal will become disconnected (stop showing output from the serial port and the ctrl-c,d and send buttons go grey). Refreshing the screen will reconnect the terminal and output will continue unless it has been stopped from the usb serial terminal.

### Description

In most cases the following message is displayed in the browser developer console:

[image]

But occasionally this message is displayed:
 
[image]

RetiredWizard, 2022-08-04:
I wouldn't be surprised if a network error is triggering this in a similar way the file manager is being disrupted in issue #6654

tannewt, 2022-08-09:
I'm going to unblock 8 on this. It'd be good for others to chime in if they see this.

RetiredWizard, 2022-08-10:
I'm not sure about the frame header error, but I can still disconnect the web workflow serial terminal with the utf-8 decode error even after applying #6742 :(

RetiredWizard, 2022-08-10:
While playing with the serial terminal I stumbled on a reliable way to reproduce the disconnect. I don't really understand what's going on with the STR variables here, but if you read a single character from the attached file it reads a string of length 4 into that single character. If you then print that character the serial terminal disconnects and you get the following message in the developer console:

     WebSocket connection to 'ws://10.0.0.206/cp/serial/' failed: Could not decode a text frame as UTF-8

Once this file has been moved to the microcontroller flash, typing these commands into the repl will cause the error:

     f = open('disconnect.txt')
     a = f.read(1)
     print(a)

I'm thinking that when there is a communications error it's possible to mangle the character data being sent via the web socket and if you end up with a malformed character similar to the one in the attached file, attempting to display it over the serial terminal causes the disconnect. I created the attached disconnect file when attempting to type data into a file using the serial terminal.

[disconnect.txt](https://github.com/adafruit/circuitpython/files/9303966/disconnect.txt)

tannewt, 2022-08-10:
I believe the issue is that the disconnect.txt file doesn't contain utf-8. When you are opening the file, you are asserting that the file is text in utf-8. CP isn't catching it is invalid. Instead it is giving you a four byte character which websocket faithfully writes out. The browser is the first to notice that it is invalid utf8. (bytes 2-4 should be 10xxxxxx).

Here is what CPython gives:

```
>>> f = open("/home/tannewt/Downloads/disconnect.txt")
>>> c = f.read(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf7 in position 0: invalid start byte
```

RetiredWizard, 2022-08-10:
I'm building an ESP32-S3 image just to confirm, but I haven't been able to generate the original disconnect from the CODE/REPL listed above on the C3 after #6742 was applied, so that PR does appear to have resolved at least part of this issue.

RetiredWizard, 2022-08-10:
Yep, the original CODE/REPL program no longer causes a disconnect after #6742. Do you want to let that PR close this and open a new one for the UTF-8 issue or just keep this one going?

tannewt, 2022-08-11:
Ya, lets open another for the UTF-8 checking. I think the PR will close this.

PR #6742 (2022-08-13): Retry send if only some bytes sent
Fixes #6654 and fixes #6689
commit 0b098f50 (2022-08-13): Retry send if only some bytes sent

Fixes #6654 and fixes #6689
closed 2022-08-13 by MicroDev1