adafruit/circuitpython #6692: circuitpython-ulab's arctan2 function returns an 1-element array instead of a float

thetazero, 2022-08-03:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.2-12-ga50b0580b-dirty on 2022-02-16; PyCubedMini with samd51J20
```

### Code/REPL

```python
from ulab import numpy as np
np.arctan2(3,1)
```

### Behavior

```python
array([1.24905], dtype=float32)
```

### Description

- Should just return a float, not an array. 
- In numpy it returns a float as expected

### Additional information

Happy to help if someone points me in the right direction.

jepler, 2022-08-03:
I verified the difference.  The implementation is in https://github.com/v923z/micropython-ulab/blob/1347694672dc76ae2d6d74fd2c76cccab7dfc311/code/numpy/vector.c#L242 so @v923z might be able to give you some guidance how to proceed.

v923z, 2022-08-04:
Thanks for pointing out the glitch! I know where the error is (universal functions with a single argument treat scalars correctly, `arctan2`, on the other hand, needed a different implementation, since one has to loop through two arrays at the same time), and the fix is trivial. I can pull it off later today.

v923z, 2022-08-04:
Fixed in https://github.com/v923z/micropython-ulab/pull/546.

thetazero, 2022-08-04:
Thanks @v923z!

jepler, 2022-08-04:
This fix will need to be incorporated into CircuitPython as well. Re-opening the issue so we can continue to track it.

PR #6939 (2022-09-23): Update ulab to 5.1.1
Closes #6692
commit 52a2c5f6 (2022-09-23): Update ulab to 5.1.1

Closes: #6692
closed 2022-09-23 by MicroDev1