adafruit/circuitpython #6703: TItle bar does not update after hard reset, from bad password to good password

dhalbert, 2022-08-05:
I connected, then changed the wifi password to be wrong, did a hard reset. I see "Authentication Failure" after a 15-second or so delay. Then I changed the password back to be correct, and did another hard reset. On the device display (Feather ESP32S2 TFT), the IP address is shown, but the terminal title bar (running `tio`) said "No IP", and it did not show the `code.py output:`. This is a little bizarre because it should have come up properly and printed the usual initial text. Instead tio just said "Connected" and idid not print anything. This seems like some kind of timing problem.

From https://github.com/adafruit/circuitpython/pull/6698#pullrequestreview-1063589464

tannewt, 2022-08-08:
Did the terminal title bar should "Auth failure" correctly? I suspect the issue may be that USB is coming up after the newest title bar updates have been sent since the display is connected from the get go.

I'll add a title bar update request when the dtr line is set over USB CDC. That way we'll send it again.

dhalbert, 2022-08-09:
> Did the terminal title bar should "Auth failure" correctly?

Right, exactly, I think your diagnosis is correct.

PR #6715 (2022-08-09): Request a title bar update on USB cdc connect
This ensures that the USB serial connection gets a copy of the
latest title bar.

Fixes #6703
commit 401c2697 (2022-08-09): Request a title bar update on USB cdc connect

This ensures that the USB serial connection gets a copy of the
latest title bar.

Fixes #6703
closed 2022-08-09 by dhalbert