adafruit/circuitpython #6707: Ctrl-C from Web Workflow terminal doesn't work if microcontroller is busy

RetiredWizard, 2022-08-06:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-102-g01f252d12-dirty on 2022-08-06; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
```

### Code/REPL

```python
import time
while True:
    time.sleep(1)
    print(".",end="")
```

### Behavior

If you connect to microcontroller over the USB serial port this program can be interrupted by pressing ctrl-C:

     ...Traceback (most recent call last):
       File "code.py", line 3, in <module>
     KeyboardInterrupt:

     Code done running.

However if you bring up the Web Workflow serial terminal when this code is running, pressing the Ctrl C or Ctrl D button does not interrupt the program. Pressing the Ctrl C button rapidly for 20 or more times will cause the serial terminal to become disconnected (no longer echo the serial output) and the board to become generally unstable.

jerryneedell, 2022-08-08:
I was just about to file a similar issue. 
On a Feather ESP32S3 running 8.0.0-alpha.1-102-g01f252d12
If I start a program executing (import foo)  vi the web workflow serial terminal then I am unable to terminate the program with a control-C.
I have to RESET the board then reconnect the serial terminal to get back to the REPL.

closed 2022-08-10 by dhalbert

jerryneedell, 2022-08-10:
Tested on eather_esp32s3_4mbflash_2mbpsram -- worked fine. Thank you!

PR #6729 (2022-08-10): Read websocket in background to look for ctrl-c
Otherwise busy Python code that isn't reading input characters
won't be interruptible.

Fixes #6707
commit 51b65cbe (2022-08-10): Read websocket in background to look for ctrl-c

Otherwise busy Python code that isn't reading input characters
won't be interruptible.

Fixes #6707