adafruit/circuitpython #6710: Displayio group not properly moved after import ?

Neradoc, 2022-08-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-103-g0ed84486f on 2022-08-08; Adafruit FunHouse with ESP32S2
7.3.2
```

### Code/REPL

```python
"""
In test_mod.py
"""
import displayio

main = displayio.Group()
sub = displayio.Group()

main.append(sub)
print("In module, is sub in main ? Expecting True:", sub in main)

"""
In code.py
"""
import displayio
from test_mod import main, sub

print("In code.py, is sub in main ? Expecting True:", sub in main)
print("Is main empty though ?", len(main) == 0)
print("Is the thing in main sub ?", main[0] == sub)

print("main[0] @", id(main[0]))
print("    sub @", id(sub))

print("""
Pratical issue: sub not in main (see above),
but believes it is in a group.
""")
group = displayio.Group()
group.append(sub)
```

### Behavior

```
code.py output:
In module, is sub in main ? Expecting True: True
In code.py, is sub in main ? Expecting True: False
Is main empty though ? False
Is the thing in main sub ? False
main[0] @ 1071170544
    sub @ 1073216608

Pratical issue: sub not in main (see above),
but believes it is in a group.

Traceback (most recent call last):
  File "code.py", line 16, in <module>
ValueError: Layer already in a group
```

### Description

When a displayio Group is added to another Group in a module _during import_, it is no longer the same object after import. It looks like a consequence of the long lived memory optimization ? Like the group is copied to the end of the heap, but the pointer in the parent group isn't updated ? I don't know, I might have that wrong.

- This doesn't happen if the code is all in code.py.
- This doesn't happen if  `main.append(sub)` is done from a function called after import.
- This doesn't seem to happen with regular lists.
- Didn't test other ports/boards.

This was reported on the forums https://forums.adafruit.com/viewtopic.php?f=60&t=193448

dhalbert, 2022-08-08:
Do you think this might be a manifestation of this bug?
https://github.com/adafruit/circuitpython/issues/2687

Neradoc, 2022-08-09:
Oh ok it's not special to displayio Groups. I'm surprised I didn't encounter it before.

dhalbert, 2023-10-27:
Fixed by https://github.com/adafruit/circuitpython/pull/8281, which removed long-lived allocations, and so removed objects moving around after creation.

closed 2023-10-27 by dhalbert