adafruit/circuitpython #6717: Feather Huzzah32 LED on dimly when not in use

RetiredWizard, 2022-08-09:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-102-g01f252d12-dirty on 2022-08-08; Adafruit Feather HUZZAH32 with ESP32
```

### Code/REPL

```python
import board,time,digitalio

#1.6 volts on D13
time.sleep(5)

led=digitalio.DigitalInOut(board.D13)
#0 volts on D13
time.sleep(5)

led.deinit()
#1.6 volts on D13
```

### Behavior

The voltage on pin D13 switches between 1.6V (when D13 isn't in use by digitalio) and zero volts as noted in the comments. 1.6 volts causes the LED on the Feather to be turned on.

RetiredWizard, 2022-08-09:
I'm thinking this has to do with the ESP32 floating GPIO high to avoid neopixel problems. There's no neopixel on this board though.

tannewt, 2022-08-09:
Yup, agree. I just fixed this for feather s2 tft as well.

In `board.c`:
```c
bool espressif_board_reset_pin_number(gpio_num_t pin_number) {
    // Pull LED down on reset rather than the default up
    if (pin_number == 13) {
        gpio_config_t cfg = {
            .pin_bit_mask = BIT64(pin_number),
            .mode = GPIO_MODE_DISABLE,
            .pull_up_en = false,
            .pull_down_en = true,
            .intr_type = GPIO_INTR_DISABLE,
        };
        gpio_config(&cfg);
        return true;
    }
    return false;
}
```

jerryneedell, 2022-08-09:
Is this also needed on the adafruit_feather_esp32s3_4mbflash_2mbpsram?
I see that the D13 LED is lit.

dhalbert, 2022-08-09:
This is probably a general problem on all Espressif boards with `board.LED` or equivalent.

PR #6753 (2022-08-13): Audit all boards with LED pins defined
Add them as MICROPY_HW_LED_STATUS so that we can share reset code
for them. They aren't actually used for the status if another RGB
option is available. (But maybe they should be.)

Fixes #6717
commit df1435ae (2022-08-13): Audit all boards with LED pins defined

Add them as MICROPY_HW_LED_STATUS so that we can share reset code
for them. They aren't actually used for the status if another RGB
option is available. (But maybe they should be.)

Fixes #6717
closed 2022-08-13 by MicroDev1