adafruit/circuitpython #6723: Web Workflow File Manager Missing Icons on MacOS

makermelissa, 2022-08-09:
### CircuitPython version

```python
Absolute latest from 8/8/22
```

### Code/REPL

```python
N/A
```

### Behavior

When browsing files on the device with web workflow, many of the unicode characters are missing. See screenshot below.

### Description

[image]

tannewt, 2022-08-09:
@Neradoc you had ideas for this right?

Neradoc, 2022-08-10:
Yeah I replaced them with emojis, which are better supported across systems. I could try researching non-emoji alternatives that are better supported, but that seems to be an area where it's hard to get any actual information.

makermelissa, 2022-08-10:
Emojis should be fine. The main non-emoji alternative is HTML entities like here: https://www.freeformatter.com/html-entities.html

PR #6733 (2022-08-10): Use emojis as icons in the Web Workflow
Fixes #6723 
Emojis are better supported across systems than symbol code points.
ℹ️ 📁 ⬇ 📄

[image]
closed 2022-08-10 by tannewt