adafruit/circuitpython #6724: Web Workflow Add support for partial writes

makermelissa, 2022-08-09:
It would be awesome if an offset could optionally be provided to support partial writes. However, this might be better in its own function utilizing the PATCH http method as this is exactly what it's intended for.

tannewt, 2022-08-09:
Why is this interesting? Wifi is faster than BLE so I think this is less critical.

We should definitely go with range requests for it: https://developer.mozilla.org/en-US/docs/Web/HTTP/Range_requests

makermelissa, 2022-08-09:
Yeah, it's definitely less critical. You had asked I create an issue, so that's what I did. The description for range requests sounds like it is intended for reads rather than than writes, so I'm not sure that would satisfy our needs.