adafruit/circuitpython #6732: Choose if a pin state should persist through deep sleep?

maholli, 2022-08-09:
## Goal
There are many cases when a user may want to conditionally set a pin high or low and then maintain that pin state during and after deep sleep.  

A common example would be choosing whether to keep some external circuit enabled during deep sleep (like a GPS module searching for a fix). 
  - Currently, this isn't possible since pins are reset just prior to entering deep sleep (in the case of the atmel-samd port). 
  - On discord, @tannewt suggested `PinAlarm` might provide a hacky workaround since you can set `pull=True` to pull the pin in the opposite direction during sleep. This sort-of works for light sleep, but at least for the atmel-samd port, the usable pins are further reduced for deep sleep down to the few tamper pins.
    - Additionally, this workaround requires a `deinit` of the pin prior to sleep if the pin was configured as digitalio during normal operation and another pin reset upon wake. This makes it unusable for the GPS example since the module will still experience two resets are require it to begin searching for a fix from scratch. 

## To summarize
There should be a way to set flag a digitalio output pin during normal operation such that the logic level of the pin is not disrupted upon entering nor during deep sleep. 

Does anyone have high-level thoughts on possible ways to achieve this? What about maintaining pin state upon wake? I'm not even sure it's feasible to not disrupt a pin as we exit deep sleep. Regardless, if we come up with something that isn't too complex, especially across ports, I can certainly help implement it.

tannewt, 2022-08-10:
Would it be enough for the board C code implement this? Or do you want control from CircuitPython of it?

maholli, 2022-08-11:
> Would it be enough for the board C code implement this? Or do you want control from CircuitPython of it?

I imagine there will certainly be a board.c component, but the user still needs to be able to set/clear the flag from their python code. Plus I would think there also needs to be a lower-level hook that appropriately checks and exempts pins from being reset while still allowing the user to claim pins during normal use.

dhalbert, 2022-10-18:
Pin value persistence through sleep was implemented by #6907, but only on Espressif. The pin state is not necessarily preserved after waking up.