adafruit/circuitpython #6737: Use new official I2C terminology in error messages.

bergdahl, 2022-08-10:
### CircuitPython version

```python
Any
```

### Code/REPL

```python
None
```

### Behavior

While doing the latest batch of localization in Weblate I discovered the text `All I2C targets are in use` and that there was already translations for `All I2C peripherals are in use`. I was advised to open an issue by @jepler to get all messages in CP in line for the new terminology.

dhalbert, 2022-08-10:
The message in question is wrong and should not be "All I2C targets are in use", because the original message "All I2C peripherals are in use" referred to the hardware blocks on the microcontroller chip, not a standalone I2C "peripheral". Such a device was previously called an "I2C slave", then we changed it to "I2C peripheral" (which confuses it with the on-chip hardware). NXP in a recent revision of the I2C standard changed "slave" to "target" (probably to avoid this confusion). See https://www.nxp.com/docs/en/user-guide/UM10204.pdf.

Following up on the discord conversation, _mål_ (_goal_) sounds ok, since it could be a physical goal (as opposed to a conceptual goal). I see that _archery target_ is _bågskyttemål_ or _mål för bågskytte_, which is in the same vein. Same for _fotbollsmål_.

closed 2022-08-10 by dhalbert

bergdahl, 2022-08-10:
OK, so peripheral = part of the MCU and target = exaternal sensor. I have changed my translation accordingly on Weblate.

PR #6738 (2022-08-10): switch back one I2C message
Fixes #6737.

I inadvertently changed "All I2C peripherals are in use" to "All I2C targets are in use" in `I2Ctarget.c` in #6721. This switches it back. Thanks to @bergdahl and @jepler for noticing.

In this particular message, "peripheral" means the hardware on the microcontroller chip. It does not refer to an "I2C target", formerly called an "I2C peripheral". The other changes of "I2C peripheral" to "I2C target" do reflect the updated official terminology.