adafruit/circuitpython #6745: esp32-eye is flaky

jepler, 2022-08-10:
### CircuitPython version

```python
8.0.0-alpha (various)
```

### Code/REPL

```python
print("Connecting to Adafruit IO")                                              
mqtt_client = MQTT.MQTT(                                                        
    broker="io.adafruit.com",                                                   
    username=aio_username,
    password=aio_key,
    socket_pool=pool,                                                           
    ssl_context=ssl.create_default_context(),                                   
)
```

### Behavior

Connecting to Adafruit IO
Traceback (most recent call last):
  File "<stdin>", line 55, in <module>
TypeError: function got multiple values for argument 'socket_pool'

### Description

The exact error varies with the order of imports, order of object creation, etc, but usually seems to boil down to getting one object confused for another in an impossible way.

The same source files and libs on an Adafruit Feather ESP32 V2 do not exhibit the same problem. However, the same file will give the same message on two different ESP32-EYE boards, and it's also consistent from run to run. (V2 feather has a different amount of PSRAM, a different PSRAM chip, and connects to different pins on the CPU)

This is with web workflow enabled and configured in CIRCUITPY/.env

### Additional information

Things To Investigate
 * sdkconfig differences
 * possible problems with the physical board
   * out of spec psram (voltage, speed) or other parts
   * check for any obviously counterfeit components

Separately, should wire up a camera to a V2 feather and verify the esp32-camera library works on esp32. I initially thought the problem might occur when the camera was used, but on further investigation this does not seem to be the case.

DavePutz, 2023-02-06:
I don't have an ESP32-EYE board; but tested adafruit-circuitpython-espressif_esp32_eye-en_US-8.0.0-rc.2.bin on a similar board (AI-Thinker ESP32-CAM). It ran successful MQTT wifi tests for over an hour on 3 separate runs.

dhalbert, 2025-02-01:
Closing this for now with no further info.

closed 2025-02-01 by dhalbert