adafruit/circuitpython #6746: STM32: Revisit F1 support

PinkFreud, 2022-08-11:
Re: #3192, #3069

Reading through the above two reports, I can see that it was decided that the F1 has too little RAM and/or flash storage for circuitpython.  I appear to have a variant which may have enough storage and RAM, though - the STM32F103RGT6, on board the [M24SR-DISCOVERY](https://www.st.com/en/evaluation-tools/m24sr-discovery.html).  According to the docs for this board, it has 1M of storage, and specs for the STM32F103RGT6 indicate 96k RAM.  While these aren't phenomenal specs, they sound like they may be usable enough for circuitpython.  Adding support for the on-board display and the M24SR NFC device will be an extra challenge, but it'd be interesting to see F1 support first.

Is this doable?  Or am I barking up the wrong tree?

dhalbert, 2022-08-11:
As you can see from #3192 and #3069, there was still a lot of work to do with the F1's, because there are idiosyncratic differences between F1 and F4. 96kB is marginal for a good CircuitPython experience, and with RP2040 and other chips available with much more RAM, it's not clear it's worth it for someone to spend time on this. Cheaper, more capable, and more readily available chips are available. Certainly don't do it just because you have the dev board; it's far less trouble to just buy a better dev board.

hierophect, 2022-08-15:
Noticed this while dropping by - I spent a bit of work on this because F1s are so highly valued by some dev circles (especially in China), but ultimately abandoned it because the ecosystem of F1 boards at the specs Circuitpython needs is very meagre. I was in a situation where the only dev boards I could find were not real F1s at all but rather unauthorized clones - and when I had problems setting them up, it felt like there was no certainty whether it was my code or some knockoff quirk.

That said I had almost completely finished the setup for the project, it just wasn't working due to some collection of clock problems. If you are determined to pick this up, I can pass along that code. It'd be out of date with the current STM32 port structure now, though.