adafruit/circuitpython #6752: Reading invalid utf-8 character from file does not produce error

RetiredWizard, 2022-08-11:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-127-gf1053fb96 on 2022-08-11; Adafruit QT Py ESP32C3 with ESP32-C3FN4
```

### Code/REPL

```python
file = open('disconnect.txt')
badchar = file.read(1)
```

### Behavior

The attached disconnect.txt contains a mangled character created through a network error. I haven't been able to figure out how to recreate it programmatically but perhaps someone with a better understanding of character sets and CP string formats could.

[disconnect.txt](https://github.com/adafruit/circuitpython/files/9311941/disconnect.txt)

When this file is opened and a single character is read into badchar, an invalid utf-8 character of length 4 and value '\x07\n\r\x08' is loaded without displaying an error.

Attempting to obtain the ord(badchar) results in the following error:

     Traceback` (most recent call last):
       File "<stdin>", line 1, in <module>
     TypeError: ord() expected a character, but string of length 4 found

Here is what CPython gives when attempting to read in the same character:

     >>> f = open("/home/tannewt/Downloads/disconnect.txt")
     >>> c = f.read(1)
     Traceback (most recent call last):
       File "<stdin>", line 1, in <module>
       File "/usr/lib/python3.10/codecs.py", line 322, in decode
         (result, consumed) = self._buffer_decode(data, self.errors, final)
     UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf7 in position 0: invalid start byte

jepler, 2022-08-11:
If you can please check the micropython differences from python document to see if this is a documented difference. I strongly suspect the behavior is the same in mp and if it's not documented there I'd reach out to them.

RetiredWizard, 2022-08-11:
I took a look at the differences documents but I'm afraid a lot of it's over my head. Regardless, you're correct, I tested the file out on micropython 1.19.1 and it behaved exactly the same. I'm fine with this falling back as a MP issue, 

it's probably not even that big an issue, my only interest in it was that the chrome browser disconnects the web workflow serial terminal when it gets one of these characters but I think most of the ways the web workflow was creating them are being shut down so we probably won't see this much going forward.

Thanks.

jepler, 2022-08-12:
Hi, this second issue (printing a string created in an unusual way in the repl causes a disconnect) seems like it's worth considering in its own right.

For background, CircuitPython and micropython assume that any file opened in text mode (the default opening mode) is actually a valid text file in the utf-8 encoding. Of course, it's easy to create a file in byte mode that is not valid utf-8 encoding, and there are lots of examples of this -- .mpy files are one that would exist on a huge fraction of all CircuitPython devices.

Being light on validation as a trade-off for speed & simplicity, micropython doesn't validate during `file.read()` that the bytes are valid as UTF-8, so the in-memory string ends up having contents that are also not valid. Actually performing operations on this string then does invalid things.  For instance, printing it with `print()` puts an invalid UTF-8 byte sequence in the output.

Terminal programs, due to their long heritage of dealing with people who accidentally `cat` binary files, tend to deal gracefully enough with this, possibly by showing the unicode "replacement character" � instead. But the web repl, as reported here, crashes. And that's potentially worth dealing with by improving robustness.

The other reason I suggested checking micropython is that if it's not a documented difference, it's probably worth raising an issue with them. They may choose to document it as a difference or to fix it and either way it helps improve both projects.

jepler, 2022-08-12:
I also want to add that I wish I had not written my first comment in this issue discussion the way I did, because it reads as being dismissive of the issue you reported (excuse: I was a passenger in a car and bored but also armed with just an onscreen keyboard and no circuitpython/micropython)

Instead, I should have taken the time with my comment and have made it clearer that I wanted to discover more about where it would be most valuable to address the problem, as a circuitpython problem, or as a micropython problem (or, which I didn't even consider at the time, as a web workflow problem).

RetiredWizard, 2022-08-12:
Please don't worry about your comment, I been reading electronic messaging from the literal beginning of electronic messaging and have long since learned not to place any subtle emotional tone on someone else's message :).

I really do appreciate your detailed explanation here, it has increased my understanding of the situation greatly. These unicode characters get stuck in my head probably because the idea of a character being 8 bits is almost as core an idea for me as that everything on computers is ones and zeros.

I do think your original suggestions hold and that closing this issue makes sense. Your explanation describing valid files containing invalid utf-8 characters does also suggest that we should open an issue regarding the possible display of those files over the serial terminal (interestingly, I believe the web  workflow editor does handle them okay).

And while I don't think I verse enough with the terminology and PEP documents referenced to determine if the Micropython difference document identifies this or not, I'll probably go ahead and open a Micropython forum question and asking if it's a documented difference and depending on the answer opening an issue there.

Thanks again :D

RetiredWizard, 2022-08-12:
Just saw your links to the Micropython issues :).

closed 2022-08-15 by tannewt

jimmo, 2022-08-22:
Thanks for reporting this @RetiredWizard -- see https://github.com/micropython/micropython/pull/9083 for a proposed fix in MicroPython.

PR #6754 (2022-08-15): When reading data from a file into a str, check if it's utf-8
Otherwise, weird stuff can happen down the line when it is print()ed, especially as it can break the webrepl of circuitpython. Closes #6752.

This needs formal tests, not just the manual testing I did.