adafruit/circuitpython #6755: FunHouse Boot Loop

jerryneedell, 2022-08-12:
### CircuitPython version

```python
using https://adafruit-circuit-python.s3.amazonaws.com/bin/adafruit_funhouse/en_US/adafruit-circuitpython-adafruit_funhouse-en_US-20220811-f9b9f55.uf2

Cannot get to boot_out.txt
```

### Code/REPL

```python
Cannot get to REPL
```

### Behavior

After loafing the .uf2, the board continually boots and reboots -- I can read the screen fast enough to get any information. Not CIRCUITPY drive and the USP bot keeps resetting.

dhalbert, 2022-08-12:
Reproduced this by:
```sh
esptool.py -p /dev/ttyACM0 -b 460800 erase_flash
esptool.py -p /dev/ttyACM0 -b 460800 --before=default_reset --after=no_reset write_flash 0x0000 adafruit-funhouse-factory-reset-and-bootloader.bin
cp adafruit-circuitpython-adafruit_funhouse-en_US-20220811-f9b9f55.uf2 /media/halbert/HOUSEBOOT/
```
the "factory reset" file is the one from the guide

jerryneedell, 2022-08-12:
I installed the bootloader (after erasing flash ) by loading an arduino sketch (blink)

jerryneedell, 2022-08-13:
The current main seems to work OK on the Funhouse now -- including web workflow

Board:
[Adafruit FunHouse](https://circuitpython.org/board/adafruit_funhouse/)
Version:
8.0.0-alpha.1-135-gfa3e265bf
Hostname:
[cpy-09c70c](http://cpy-09c70c.local/)
IP:
[10.0.0.47](http://10.0.0.47/)

MicroDev1, 2022-08-13:
The current main works fine for me too.

closed 2022-08-13 by MicroDev1