adafruit/circuitpython #6756: ESP32-S2 w/ TFT screen flickers upon startup

TrevKnows, 2022-08-12:
### CircuitPython version

```python
Issue occurring using absolute build Adafruit CircuitPython 8.0.0-alpha1 using:
adafruit-circuitpython-adafruit_feather_esp32s2_tft-en_US-20220811-f9b9f55.uf2

to build:
adafruit-circuitpython-adafruit_feather_esp32s2_tft-en_US-20220810-a1d48e9.uf2

This build works:
adafruit-circuitpython-adafruit_feather_esp32s2_tft-en_US-20220810-08e3133.uf2
```

### Code/REPL

```python
There was no code added to the code.py. Also, unable to check REPL because the board isn't being recognized.
```

### Behavior

My computer does not recognize the board. 
On startup or when the board is powered on, some text appears on the TFT, flickers; then the screen goes black. This is repeatedly happening.

tannewt, 2022-08-15:
@dhalbert You were gonna look into this right?

ladyada, 2022-08-17:
(this is bootloop issue, which i could repro!)

tannewt, 2022-08-17:
Ok, I found an issue with the websocket ringbuf. It does a % which crashes when size is 0 before init. Fix coming soon.

PR #6778 (2022-08-18): Guard against workflow background happening early
Fixes #6756 because the ringbuf size is zero and the empty check
does % which leads to division by zero error.
commit 704ecc34 (2022-08-18): Guard against workflow background happening early

Fixes #6756 because the ringbuf size is zero and the empty check
does % which leads to division by zero error.
closed 2022-08-18 by dhalbert