adafruit/circuitpython #6761: `CIRCUITPY_BLEIO = 0` yields `error: implicit declaration of function 'bleio_reset'`

rgbpcb, 2022-08-14:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-31-gc1b00442a on 2022-07-02; stm32f411ce-blackpill with STM32F411CE
```

### Code/REPL

```python
CIRCUITPY_BLEIO = 0
```

### Behavior

When I add `CIRCUITPY_BLEIO = 0` to `mkconfigboard.mk` (and similar lines for other libraries, in order to save space) and run `make`, I get:
```
GEN build-myboard/genhdr/moduledefs.h
QSTR updated
../../main.c: In function 'reset_devices':
../../main.c:127:5: error: implicit declaration of function 'bleio_reset'; did you mean 'atexit_reset'? [-Werror=implicit-function-declaration]
  127 |     bleio_reset();
      |     ^~~~~~~~~~~
      |     atexit_reset
../../main.c:127:5: error: nested extern declaration of 'bleio_reset' [-Werror=nested-externs]
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:62: build-myboard/main.o] Error 1
```

MicroDev1, 2022-08-14:
Hi, @rgbpcb. Run `make BOARD=stm32f411ce-blackpill clean` before the next build.

jepler, 2022-08-14:
Based on looking near the lines in question...
```c
static void reset_devices(void) {
    #if CIRCUITPY_BLEIO_HCI
    bleio_reset();
    #endif
}
```
It looks like it is an error to disable BLEIO without also disabling BLEIO_HCI. Many invalid configurations are possible and in general no effort is made to detect them.

rgbpcb, 2022-08-15:
Thank you, `CIRCUITPY_BLEIO_HCI = 0` solved the problem.

> Many invalid configurations are possible and in general no effort is made to detect them.

Do I understand correctly that no effort _should_ be made is some cases? For example, if a user _explicitly_ says 
```
CIRCUITPY_BLEIO = 0
CIRCUITPY_BLEIO_HCI = 1
```
then it isn't clear whether they want
```
CIRCUITPY_BLEIO = 1
CIRCUITPY_BLEIO_HCI = 1
```
or
```
CIRCUITPY_BLEIO = 0
CIRCUITPY_BLEIO_HCI = 0
```

On the other hand, if the user doesn't explicitly say `CIRCUITPY_BLEIO_HCI = 1`, then maybe `CIRCUITPY_BLEIO = 0` should also cause `CIRCUITPY_BLEIO_HCI = 0`. The `?=` operator (initialize variable if it doesn't exist yet) is use extensively in CircuitPython makefiles. Particularly [py/circuitpy_mpconfig.mk ](https://github.com/adafruit/circuitpython/blob/main/py/circuitpy_mpconfig.mk) might have something like
```
ifeq ($(CIRCUITPY_BLEIO),0)
CIRCUITPY_BLEIO_HCI ?= $(CIRCUITPY_BLEIO)
endif
```
(note the `?=` operator which preserves `CIRCUITPY_BLEIO_HCI = 1` if the user explicitly wanted that).

**On a second thought, dependencies might change, and keeping track of them imperfectly might cause code to behave wrong, which is worse than a compile error.**

Maybe add a _warning_?

closed 2022-08-15 by rgbpcb

dhalbert, 2022-08-15:
It would need to be a warning. `CIRCUITPY_BLEIO_HCI` denotes the HCI co-processor implementation of `CIRCUITPY_BLEIO`. The default "on chip" implementation (e.g., nRF52840) is meant when `CIRCUITPY_BLEIO = 1` and `CIRCUITPY_BLEIO_HCI = 0`.