adafruit/circuitpython #6762: esp32_s2 pinging too fast crashes CircuitPython

bill88t, 2022-08-14:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.2-dirty on 2022-07-30; Waveshare ESP32-S2-Pico with ESP32S2
```

### Code/REPL

```python
import wifi
from ipaddress import ip_address
wifi.radio.connect(...) # insert your newclear launch codes here
while True:
    wifi.radio.ping(ip_address("1.1.1.1"))
```

### Behavior

[image]

### Description

I was implementing the ping command for ljinux using `wifi.radio.ping`.
But I forgot to add a time.sleep in the `while True` so it spammed a morbillion pings.
Did do (in and outside of ljinux) it 3 times with 100% success rate. (see Code/REPL)
It may take a few seconds till it crashes.

About the dirty build: https://github.com/adafruit/circuitpython/pull/6672, no official build yet.
It does not contain any further changes.

anecdata, 2022-08-15:
Dupe of https://github.com/adafruit/circuitpython/issues/5980

bill88t, 2022-08-15:
Sorry. I will close this issue now.

closed 2022-08-15 by bill88t