adafruit/circuitpython #6763: SD card emulator library/module

arekm, 2022-08-15:
There is a library to read sd cards but there is no library to "become" sd card to other hardware
(via small pcb adapter inserted into sd card slot of other hardware and connected to circuitpython device).

The goal would be to for, example, expose fs of hardware running circuitpython as sd card in a way so both,
circuitpython and other hardware could access it.

Some other projects for reference:
https://github.com/ZipCPU/sdspi/blob/master/bench/cpp/sdspisim.cpp
https://github.com/ZipCPU/zbasic/blob/master/sw/board/sdtest.c
https://github.com/raspberrypi/pico-examples/tree/master/spi/spi_master_slave
https://wiki.tizen.org/SDWire (but this only does switching between real sd card and sd card--usb storage chip)
https://github.com/FYSETC/FYSETC-SD-WIFI , https://github.com/FYSETC/ESPWebDAV (another switching device, this time between real sd card and esp8266)
https://www.eevblog.com/forum/microcontrollers/sd-card-emulator/ old discussion on eevblog

dhalbert, 2022-08-15:
For this we need SPI secondary (device) support, which is not implemented on any port right now.

tannewt, 2022-08-16:
> circuitpython and other hardware could access it.

Note that SD cards operate at the block level and you'll have the same issue that we do with CP and a host computer. What device are you trying to share the FS with?

arekm, 2022-08-16:
3d printer. It's a bit easier with 3d printer since most of them support remote command to release sd card (usually to remove it and also command to init it for usage). Parallel usage would be of much better but no idea if that can be done.

How CP is doing it?

tannewt, 2022-08-16:
Can your 3D printer do USB host? It might be simpler to plug the circuitpython device into it.

arekm, 2022-08-16:
Printer only exposes serial console over USB and the only storage medium is sd card. Currently using sdwire hardware (mentioned in first post) but that is just dumb switch without any programming capability.