adafruit/circuitpython #6773: microcontroller.cpu.temperature not implemented on ESP32-S3

kattni, 2022-08-17:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-09; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import microcontroller

while True:
    print(microcontroller.cpu.temperature)
```

### Behavior

This code throws the following error:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 4, in <module>
NotImplementedError:

Code done running.
```

dhalbert, 2022-08-17:
Temperature is not implemented in ESP-IDF yet for ESP32-S3. It's part of the ULP. See: https://github.com/espressif/esp-idf/issues/8672

igrr, 2022-11-11:
The temperature sensor API is available for the ESP32-S3: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/temp_sensor.html
The linked issue is related to the temperature sensor driver for the ULP coprocessor.

dhalbert, 2022-11-11:
@igrr Thanks - I will see about adding it. We are still on 4.4.3 at the moment.

closed 2022-11-12 by dhalbert