adafruit/circuitpython #6775: Console not visible: TFTFeatherWing 3.5" on UM FeatherS2

rsbohn, 2022-08-17:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-143-g2f0e209ec on 2022-08-17; FeatherS2 with ESP32S2
```

### Code/REPL

```python
import board
import displayio
from adafruit_featherwing.tft_featherwing_35 import TFTFeatherWing35

if "feathers2" in board.board_id:
    wing = TFTFeatherWing35(cs=board.D5, dc=board.D6, ts_cs=board.D21, sd_cs=board.D20)
    display = wing.display
    display.rotation=180
    ts = wing.touchscreen
else:
    raise NotImplementedError("Where am I?")
```

### Behavior

With CircuitPython 7.3.2 this code exits and I can see the REPL prompt on the TFT screen.

With the 8.0 alpha I see the status bar at the top of the screen, but the REPL console is not visible.

rsbohn, 2022-08-17:
Same build works as expected on the SH1107 128x64 OLED with this code:

```# oled SH1107 128x64
import board
import displayio
import adafruit_displayio_sh1107

displayio.release_displays()
i2c = board.I2C()
bus = displayio.I2CDisplay(i2c, device_address=0x3C)
display = adafruit_displayio_sh1107.SH1107(bus,
    width=128, height=64, rotation=0)
print("Hello OLED!")
```

DJDevon3, 2022-08-17:
I have the 3.5" TFT featherwing and UM S3 on 8.0. The console shows up fine though I'm not using the featherwing library. 

On Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-09; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3

```py
import board
import displayio
from adafruit_hx8357 import HX8357

# Display Dimensions
DISPLAY_WIDTH = 480
DISPLAY_HEIGHT = 320

# Initialize TFT Display
displayio.release_displays()
spi = board.SPI()
tft_cs = board.IO1
tft_dc = board.IO3
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, baudrate=50000000, phase=0, polarity=0)
display = HX8357(display_bus, width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT)
print("Hello World")
```
I just use fourwire. I've actually never used the featherwing library. Above code works fine for me. I suspect the issue is within the featherwing library or the way you're calling the display vs displayio fourwire. Won't really know until someone with a UM S2 and 3.5" featherwing, and uses the featherwing library steps up to confirm. My UM S3 is close but not identical.

rsbohn, 2022-08-18:
```# TFT FeatherWing 3.5"
# Using the HX8357 driver directly
import os
import board
import displayio
from adafruit_hx8357 import HX8357

# Display Dimensions
DISPLAY_WIDTH = 480
DISPLAY_HEIGHT = 320

# Initialize TFT Display
displayio.release_displays()
spi = board.SPI()
tft_cs = board.IO1
tft_dc = board.IO3
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, baudrate=50000000, phase=0, polarity=0)
display = HX8357(display_bus, width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT)
display.rotation=180
print()
print(os.uname().machine)
print(os.uname().version)
```
I get the same result: Status bar but no console text.

I realized that it's not just the REPL, I don't see anything `print()`ed on the TFT, but it does show up on the serial console (in Mu).

rsbohn, 2022-08-18:
```# where did it break?
# feathers2-en_GB-8.0.0-alpha.1.uf2 -- working
# 8.0.0-alpha.1-85-g462eb7157 on 2022-07-30 -- working
# 8.0.0-alpha.1-109-gdf52d995e on 2022-08-09 -- broken
# 8.0.0-alpha.1-96-ge0cb8ef17 on 2022-08-04 -- working
# 8.0.0-alpha.1-101-g8c10e09ba on 2022-08-05 -- working (status bar is just blinka icon)
# 8.0.0-alpha.1-108-gd95cfc3ab on 2022-08-09 -- broken :(
# 8.0.0-alpha.1-104-gf1826b054 on 2022-08-08 -- broken
# 8.0.0-alpha.1-102-g01f252d12 on 2022-08-06 -- broken ** previous build was working!
```
How do I match the 'g01f252d12` build (en_GB-20220805-8c10e09) with a pull request? The timestamp in S3 is 2022-08-06T00:47:01.000Z.

DJDevon3, 2022-08-18:
OK then it's definitely not the featherwing library and seems like an issue specific to the UM S2. Great investigative work btw.

tannewt, 2022-08-19:
> How do I match the 'g01f252d12` build (en_GB-20220805-8c10e09) with a pull request? The timestamp in S3 is 2022-08-06T00:47:01.000Z.

`g01f252d12` is `g<start of git commit> ` so... 01f252d12 was made with the merge of https://github.com/adafruit/circuitpython/pull/6698

tannewt, 2022-08-19:
I've replicated this on a Feather M4 Express. Looking into it now.

PR #6783 (2022-08-20): Fix scroll area shift with scale > 1
Fixes #6775
commit 802cff22 (2022-08-20): Fix scroll area shift with scale > 1

Fixes #6775
closed 2022-08-20 by jepler