adafruit/circuitpython #6791: Crash into the HardFault_Handler

DJDevon3, 2022-08-21:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.2 on 2022-07-20; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3

and 

Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3
```

### Code/REPL

```python
# Connect to WiFi
try:
    print("\n===============================")
    print("Connecting to WiFi...")
    pool = socketpool.SocketPool(wifi.radio)
    requests = adafruit_requests.Session(pool, ssl.create_default_context())
    wifi.radio.connect(secrets["ssid"], secrets["password"])
except (ValueError, RuntimeError) as e:
    print("Failed to get data, retrying\n", e)
    wifi.reset()
    time.sleep(60)
print("Connected!\n")
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.2 on 2022-07-20; FeatherS3 with ESP32S3

### Description

Issue documented in discord dev channel. https://discord.com/channels/327254708534116352/327298996332658690/1010913898708336681

### Additional information

Issue is worse on 8.0 beta than 7.3 because it still crashes to safe mode but without any error describing it in repl like it shows in 7.3. but issue occurs in both UF2 versions. 

after a certain period of time when i hit save in Mu the USB drive disconnects, reconnects, i pull up repl immediately, and it shows that message. usually about 2-3 minutes. i have to physically hit reset on the board, get to save my code once or twice before 2-3 minutes, and the process repeats itself. It’s a safe mode crashing loop. 

here's the current full code i'm using, including /lib and everything. https://github.com/DJDevon3/CircuitPython/tree/main/UM_ESP32-S3%20Online-Offline%20Weatherstation

DJDevon3, 2022-08-21:
Also I’m running almost identical code on another 3.5” tft featherwing except on the bluefruit sense and esp32 airlift. That one works no problem.

DJDevon3, 2022-08-27:
Lost power during a storm today. Restarted PC. The device is unrecognized in Windows. Throws an error when unplug/replug.

[image]

Intentionally didn't hit the reset button to collect info on this issue.
[image]

Hit reset on the board.
[image]

Open Mu
[image]

Load code.py manually
[image]

Pops up with another USB warning about the drive having a problem.

Open Serial Console and immediately greeted with this message.
[image]

Ctrl+D & Ctrl+C only reloads the safe mode message.  Cannot use Mu. 

Reset the board a 2nd time... works.
[image]

If I attempt to save any code from here it will crash to safe mode again, this is the entire safe mode crash loop experience.

DJDevon3, 2022-08-27:
Narrowed it down to the wifi or online portions of the code causing it. In it's offline state with no wifi code in it, it's works perfectly fine, can upload code with Mu and use REPL without issue.

DJDevon3, 2022-08-27:
Ugh this is like hitting a moving target. Backed out of everything, returned to original code.py and everything works fine. This thing is time based, I have to wait like 10 minutes, put Mu in the background, then come back to it in order to replicate.

DJDevon3, 2022-08-27:
Slimmed it down to this, saved, came back 10 minutes later, hit save again, works perfectly fine.

``` py
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
# SPDX-License-Identifier: MIT
"""DJDevon3 UM ESP32 Feather S3 Online Weatherstation"""

import board
import displayio
from adafruit_display_text import label
from adafruit_bitmap_font import bitmap_font
from adafruit_bme280 import basic as adafruit_bme280
from adafruit_hx8357 import HX8357

# 3.5" TFT Featherwing is 480x320
displayio.release_displays()
DISPLAY_WIDTH = 480
DISPLAY_HEIGHT = 320

# Initialize TFT Display
spi = board.SPI()
tft_cs = board.IO1
tft_dc = board.IO3
display_bus = displayio.FourWire(spi,
                                 command=tft_dc,
                                 chip_select=tft_cs,
                                 baudrate=50000000,
                                 phase=0,
                                 polarity=0)
display = HX8357(display_bus, width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT)

# Initialize BME280 Sensor
i2c = board.I2C()  # uses board.SCL and board.SDA
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)

# Quick Colors for Labels
text_black = 0x000000
text_blue = 0x0000FF
text_cyan = 0x00FFFF
text_gray = 0x8B8B8B
text_green = 0x00FF00
text_lightblue = 0x90C7FF
text_magenta = 0xFF00FF
text_orange = 0xFFA500
text_purple = 0x800080
text_red = 0xFF0000
text_white = 0xFFFFFF
text_yellow = 0xFFFF00

# Fonts are optional
medium_font = bitmap_font.load_font("/fonts/Arial-16.bdf")
huge_font = bitmap_font.load_font("/fonts/GoodTimesRg-Regular-121.bdf")

temp_label = label.Label(medium_font)
temp_label.anchor_point = (1.0, 1.0)
temp_label.anchored_position = (475, 145)
temp_label.scale = 2
temp_label.color = text_white

temp_data_label = label.Label(huge_font)
temp_data_label.anchor_point = (0.5, 1.0)
temp_data_label.anchored_position = (DISPLAY_WIDTH / 2, 200)
temp_data_label.scale = 1
temp_data_label.color = text_orange

temp_data_shadow = label.Label(huge_font)
temp_data_shadow.anchor_point = (0.5, 1.0)
temp_data_shadow.anchored_position = (DISPLAY_WIDTH / 2 + 2, 200 + 2)
temp_data_shadow.scale = 1
temp_data_shadow.color = text_black

humidity_label = label.Label(medium_font)
humidity_label.anchor_point = (0.0, 1.0)
humidity_label.anchored_position = (5, DISPLAY_HEIGHT - 23)
humidity_label.scale = 1
humidity_label.color = text_gray

humidity_data_label = label.Label(medium_font)
humidity_data_label.anchor_point = (0.0, 1.0)
humidity_data_label.anchored_position = (5, DISPLAY_HEIGHT)
humidity_data_label.scale = 1
humidity_data_label.color = text_orange

barometric_label = label.Label(medium_font)
barometric_label.anchor_point = (1.0, 1.0)
barometric_label.anchored_position = (470, DISPLAY_HEIGHT - 27)
barometric_label.scale = 1
barometric_label.color = text_gray

barometric_data_label = label.Label(medium_font)
barometric_data_label.anchor_point = (1.0, 1.0)
barometric_data_label.anchored_position = (470, DISPLAY_HEIGHT)
barometric_data_label.scale = 1
barometric_data_label.color = text_orange

# Load Bitmap to tile grid first (Background layer)
DiskBMP = displayio.OnDiskBitmap("/images/Astral_Fruit_8bit.bmp")
tile_grid = displayio.TileGrid(
    DiskBMP,
    pixel_shader=DiskBMP.pixel_shader,
    width=1,
    height=1,
    tile_width=DISPLAY_WIDTH,
    tile_height=DISPLAY_HEIGHT)

# Create subgroups
text_group = displayio.Group()
text_group.append(tile_grid)
temp_group = displayio.Group()
main_group = displayio.Group()

# Add subgroups to main display group
main_group.append(text_group)
main_group.append(temp_group)

# Label Display Group (foreground layer)
temp_group.append(temp_label)
temp_group.append(temp_data_shadow)
temp_group.append(temp_data_label)
text_group.append(humidity_label)
text_group.append(humidity_data_label)
text_group.append(barometric_label)
text_group.append(barometric_data_label)
display.show(main_group)

while True:
    temp_label.text = "°F"
    temperature = "{:.1f}".format(bme280.temperature * 1.8 + 32)
    temp_data_shadow.text = temperature
    temp_data_label.text = temperature
    humidity_label.text = "Humidity"
    humidity_data_label.text = "{:.1f} %".format(bme280.relative_humidity)
    barometric_label.text = "Pressure"
    barometric_data_label.text = "{:.1f}".format(bme280.pressure)
```

DJDevon3, 2022-08-29:
Ran some tests on Adafruit Feather ESP32-S2 with the same code. Issue is not present on the S2.

DJDevon3, 2022-09-14:
Updated to nightly build hoping it would sort some stuff out after UM pushed a PR for LDO pin and some other bug hunt stuff going on recently. Still getting the same issue. Hasn't given me a hard fault crash yet. After a couple minutes after hard reset Mu doesn't seem to be able to save to the board. It's hanging during the process and waits to reload forever, indefinitely. Ctrl+C or Ctrl+D unresponsive.

Adafruit CircuitPython 8.0.0-beta.0-47-gae64f9fd7 on 2022-09-14; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3

Code stopped by auto-reload. Reloading soon.

DJDevon3, 2022-09-14:
Oops, was using 7.3.3 libraries. Added in 8.x libraries and it's working better. It refuses to reconnect to wifi if using Mu to soft-reload. It'll never connect. Hit reset and keep Mu closed and it will connect to wifi every time. Can only use Mu for serial monitoring after a hard reset. As soon as you hit save, auto reload, it'll never connect again.  

Haven't hit safe mode & hard fault with new 8.x 9/14 bundle libraries yet.

dhalbert, 2022-09-30:
Based on your testing this afternoon, should we close this?

DJDevon3, 2022-10-01:
I'm hesitant to close this just yet, only did about 5 mins worth of tests which isn't enough time for this bug especially knowing others are having very similar hard faults. It'll be great if all the hard faults end up being USB hub or controller issues. I'd like to spend at least a couple days with it just in case. Also have some more tests to do on the rp2040, m4 express, and S2 as well.

DJDevon3, 2022-10-01:
Issue persists. Still crashing to safe mode and hard faulting.  Windows continually reports an error with the drive, there's some kind of issue with the way windows is handling this particular drive vs all of my other boards. It's also the only board that's crashing with hard faults.

dhalbert, 2022-10-01:
Do you have another S3 board to test?

You might just try reformatting the filesystem too, with `storage.erase_filesystem()`.

DJDevon3, 2022-10-01:
I have a lot of different boards I can test. RP2040, M0 Wifi, M4 Express, Bluefruit Sense, Pi Pico, ESP32-S2, QT Py S3.

DJDevon3, 2022-10-01:
I do actually have another S3 but it's the same exact UM Feather S3. Bought 2.  Oh I've run it through the nuking process twice to no avail.

DJDevon3, 2022-10-01:
Upon plugging in the 2nd UM FeatherS3 Windows immediately popped up the same message about there being a problem with the drive. Windows does not like this board. No errors in device manager.

Running the default code that comes with the UM FeatherS3 code.py
```py
import time, gc, os
import neopixel
import board, digitalio
import feathers3

# Create a NeoPixel instance
# Brightness of 0.3 is ample for the 1515 sized LED
pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.3, auto_write=True, pixel_order=neopixel.RGB)

# Say hello
print("\nHello from FeatherS3!")
print("------------------\n")

# Show available memory
print("Memory Info - gc.mem_free()")
print("---------------------------")
print("{} Bytes\n".format(gc.mem_free()))

flash = os.statvfs('/')
flash_size = flash[0] * flash[2]
flash_free = flash[0] * flash[3]
# Show flash size
print("Flash - os.statvfs('/')")
print("---------------------------")
print("Size: {} Bytes\nFree: {} Bytes\n".format(flash_size, flash_free))

print("Pixel Time!\n")

# Create a colour wheel index int
color_index = 0

# Turn on the power to the NeoPixel
feathers3.set_ldo2_power(True)

# Rainbow colours on the NeoPixel
while True:
    # Get the R,G,B values of the next colour
    r,g,b = feathers3.rgb_color_wheel( color_index )
    # Set the colour on the NeoPixel
    pixel[0] = ( r, g, b, 0.5)
    # Increase the wheel index
    color_index += 1

    # If the index == 255, loop it
    if color_index == 255:
        color_index = 0
        # Invert the internal LED state every half colour cycle
        feathers3.led_blink()
        
    # Sleep for 15ms so the colour cycle isn't too fast
    time.sleep(0.015)
    
```
While the first hard faulting feather is plugged in I can use the 2nd feather no problem. 2nd feather seems fine. It does take much longer than normal for it to soft-reload.  Like 5-10 seconds. It's slow to boot compared to most other boards I own.

Pasted over the same script from the first feather and it ran. Throwing some errors for missing libs because I didn't have it plugged into the tft featherwing. Plugged into the tft featherwing, rebooted and it immediately crashed into the hard fault handler. Removed it from the featherwing. Removed all display related stuff from the script. Reset, ran again, still crashing to hard fault handler. Saw that I didn't unplug the crashing feather, transferred images/icons/font/libs to 2nd feather, then unplugged crashing feather... 

Same scripts and everything duplicated to the 2nd UM FeatherS3 works fine. Can soft reload no problem. REPL is responsive nothing out of the ordinary. Wait 5 minutes, hit save in Mu again on the 2nd S3 and it crashes to the HardFault_Hander. There is a problem with UM FeatherS3's.

Will try harder to narrow down the problem.

dhalbert, 2022-10-01:
You might check with @UnexpectedMaker at their site about this.

DJDevon3, 2022-10-01:
This code does not crash. Unused imports compared to crashing script are:
import terminalio
import adafruit_imageload
import ssl
import adafruit_requests
from adafruit_display_text import label
from adafruit_bitmap_font import bitmap_font

pretty sure I can remove all the display related stuff and it'll be fine.  

```py
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
# SPDX-License-Identifier: MIT
"""DJDevon3 UM ESP32 Feather S3 Online Weatherstation"""
import gc
import supervisor
import time
import board
import feathers3
import displayio
import digitalio
import adafruit_sdcard
import storage
import wifi
import socketpool
import adafruit_requests
from adafruit_hx8357 import HX8357

# USB Power Sensing
usb_sense = supervisor.runtime.serial_connected

# 3.5" TFT Featherwing is 480x320
displayio.release_displays()
DISPLAY_WIDTH = 480
DISPLAY_HEIGHT = 320

# Initialize WiFi Pool (This should always be near the top of a script!)
# anecdata: you only want to do this once early in your code pool.
# Highlander voice: "There can be only one pool"
pool = socketpool.SocketPool(wifi.radio)

# Time between weather updates
# 900 = 15 mins, 1800 = 30 mins, 3600 = 1 hour
sleep_time = 900

# Initialize TFT Display
spi = board.SPI()
tft_cs = board.IO1
tft_dc = board.IO3
display_bus = displayio.FourWire(spi,
                                 command=tft_dc,
                                 chip_select=tft_cs)
display = HX8357(display_bus, width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT)

# Initialize SDCard on TFT Featherwing
cs = digitalio.DigitalInOut(board.IO33)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
virtual_root = "/sd"
storage.mount(vfs, virtual_root)

Bat_S3 = feathers3.get_battery_voltage()
# print(Bat_S3)

try:
    from secrets import secrets
except ImportError:
    print("Secrets File Import Error")
    raise

if sleep_time < 60:
    sleep_time_conversion = "seconds"
    sleep_int = sleep_time
elif 60 <= sleep_time < 3600:
    sleep_int = sleep_time / 60
    sleep_time_conversion = "minutes"
elif 3600 <= sleep_time < 86400:
    sleep_int = sleep_time / 60 / 60
    sleep_time_conversion = "hours"
else:
    sleep_int = sleep_time / 60 / 60 / 24
    sleep_time_conversion = "days"

def _format_datetime(datetime):
    return "{:02}/{:02}/{} {:02}:{:02}:{:02}".format(
        datetime.tm_mon,
        datetime.tm_mday,
        datetime.tm_year,
        datetime.tm_hour,
        datetime.tm_min,
        datetime.tm_sec,
    )

def _format_date(datetime):
    return "{:02}/{:02}/{:02}".format(
        datetime.tm_year,
        datetime.tm_mon,
        datetime.tm_mday,
    )

def _format_time(datetime):
    return "{:02}:{:02}".format(
        datetime.tm_hour,
        datetime.tm_min,
        # datetime.tm_sec,
    )

while True:
    gc.collect()

    # Changes battery voltage color depending on charge level
    print("USB Sense: ", usb_sense)
    if usb_sense:
        print("USB connected")

    if not usb_sense:
        print("USB not connected")

    time.sleep(sleep_time)

```

DJDevon3, 2022-10-01:
Tried narrowing it down more by removing all display related stuff.
Unused imports compared to original are:
import board
import displayio
import digitalio
import terminalio
import storage

```py
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
# SPDX-License-Identifier: MIT
"""DJDevon3 UM ESP32 Feather S3 Online Weatherstation"""
import gc
import supervisor
import time
import feathers3
import ssl
import wifi
import socketpool
import adafruit_requests

# USB Power Sensing
usb_sense = supervisor.runtime.serial_connected

# Initialize WiFi Pool (This should always be near the top of a script!)
# anecdata: you only want to do this once early in your code pool.
# Highlander voice: "There can be only one pool"
pool = socketpool.SocketPool(wifi.radio)

Bat_S3 = feathers3.get_battery_voltage()
# print(Bat_S3)

try:
    from secrets import secrets
except ImportError:
    print("Secrets File Import Error")
    raise

# Fill OpenWeather 2.5 API with token data
# OpenWeather free account & token are required
timezone = secrets['timezone']
tz_offset_seconds = int(secrets['timezone_offset'])

# OpenWeather 2.5 Free API
MAP_SOURCE = "https://tile.openweathermap.org/map"
MAP_SOURCE += "/precipitation_new"
MAP_SOURCE += "/9"
MAP_SOURCE += "/" + secrets['openweather_lat']
MAP_SOURCE += "/" + secrets['openweather_lon']
MAP_SOURCE += "/.png"
MAP_SOURCE += "&appid=" + secrets['openweather_token']

DATA_SOURCE = "https://api.openweathermap.org/data/2.5/onecall?"
DATA_SOURCE += "lat=" + secrets['openweather_lat']
DATA_SOURCE += "&lon=" + secrets['openweather_lon']
DATA_SOURCE += "&timezone=" + timezone
DATA_SOURCE += "&timezone_offset=" + str(tz_offset_seconds)
DATA_SOURCE += "&exclude=hourly,daily"
DATA_SOURCE += "&appid=" + secrets['openweather_token']
DATA_SOURCE += "&units=imperial"

def _format_datetime(datetime):
    return "{:02}/{:02}/{} {:02}:{:02}:{:02}".format(
        datetime.tm_mon,
        datetime.tm_mday,
        datetime.tm_year,
        datetime.tm_hour,
        datetime.tm_min,
        datetime.tm_sec,
    )
[... 96 lines cut ...]
    response = None

    time.sleep(sleep_time)
```
This produces the most interesting result yet. An error I've never seen before. Refuses to connect to wifi, maybe I missed something?

```py
===============================
Connecting to WiFi...
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
Retrying in 10 seconds```

The error handler is catching a 205... what's a 205?

DJDevon3, 2022-10-01:
Narrowed down the 205 as much as possible
```py
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
# SPDX-License-Identifier: MIT
# UM ESP32 FeatherS3

import wifi
import socketpool

# Initialize WiFi Pool 
pool = socketpool.SocketPool(wifi.radio)

try:
    from secrets import secrets
    timezone = secrets['timezone']
    tz_offset_seconds = int(secrets['timezone_offset'])
except ImportError:
    print("Secrets File Import Error")
    raise

# Connect to Wi-Fi
print("\n===============================")
print("Connecting to WiFi...")
while not wifi.radio.ipv4_address:
    try:
        wifi.radio.connect(secrets['ssid'], secrets['password'])
    except ConnectionError as e:
        print("Connection Error:", e)
        print("Retrying in 10 seconds")
        
# With the 205 Error it never reaches below this point
print("Connected!\n")
```
Oh duh because it's being caught by the wifi error handler the 205 error is internally wifi library related. Well at least I was able to narrow that one down.

```py
===============================
Connecting to WiFi...
Connection Error: Unknown failure 205
Retrying in 10 seconds
Connection Error: Unknown failure 205
```

UnexpectedMaker, 2022-10-01:
I'm kinda confused by this issue - is this about hard faults or wifi issues?
Re the hard faults - cant see that having anything to do with my boards. Sounds like a combo of Mu and Windows.
Does it hard fault when you don't use Mu? shut it down, use Notepad?

UnexpectedMaker, 2022-10-01:
Re the Wifi - your last code example - does that work on the QTPy S3?
Can you take a photo of your board and wiring and everything for me?

anecdata, 2022-10-01:
`205` is a quite unhelpful error code that arises from time to time:
`CONNECTION_FAIL`  "Espressif-specific Wi-Fi reason code: the connection to the AP has failed."
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html

DJDevon3, 2022-10-02:
Yes the small sketch works on the Qt Py S3.

Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-09; Adafruit QT Py ESP32-S3 no psram with ESP32S3
Board ID:adafruit_qtpy_esp32s3_nopsram

```py
# secrets.py
secrets = {
    "ssid": "Your SSID",
    "password": "Your Wifi Password",
    "timezone": "America/New_York",  # Check http://worldtimeapi.org/timezones
    "aio_username": "AdafruitIO_Username",
    "aio_key": "AdafruitIO_Key",
    "timezone_offset": "-14400",  # New York timezone. offset in seconds plus or minus
    "openweather_token" : "long_token_here",
    "openweather_lat" : "27.44", #default lat/lon is adafruit industries
    "openweather_lon" : "-80.32",
}
```

```py
# code.py
import wifi
import socketpool

# Initialize WiFi Pool 
pool = socketpool.SocketPool(wifi.radio)

try:
    from secrets import secrets
    timezone = secrets['timezone']
    tz_offset_seconds = int(secrets['timezone_offset'])
except ImportError:
    print("Secrets File Import Error")
    raise

# Connect to Wi-Fi
print("\n===============================")
print("Connecting to WiFi...")
while not wifi.radio.ipv4_address:
    try:
        wifi.radio.connect(secrets['ssid'], secrets['password'])
    except ConnectionError as e:
        print("Connection Error:", e)
        print("Retrying in 10 seconds")
        
# With the 205 Error it never reaches below this point
print("Connected!\n")
```
Connected first attempted, consecutive reloads also connected.
```py
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

===============================
Connecting to WiFi...
Connected!

Code done running.
```
Pictures of boards and setup soon to follow.

DJDevon3, 2022-10-02:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

UnexpectedMaker, 2022-10-02:
Interesting that worked for months for you - I know not your choice, but you have the hi speed signal wires of the screen sitting at the antenna - that's going to cause grief.  Plus it's inside a case, reducing the antenna gain even more, and you have an I2C device hanging off the back next to the antenna - it's a worst case setup there - I'm surprised that ever worked for you at all ;)

That said, if it was working - and now it's not...   I'm still perplexed as to what the reported problem is about.

Wifi was working fine and now isn't?  
Never had any hard faults but now there is?

If the HW hasn't changed though all of that...  and there have only been SW and firmware level changes - I'm not sure where I'm supposed to giving input on this?

UnexpectedMaker, 2022-10-02:
It looks like your AP is just failing to assign the FS3 an IP address. 
Have you tried rebooting the router?
Have you tried assigning a static IP in the DHCP list for the FS3?

DJDevon3, 2022-10-02:
Oh I said it was in that configuration since the day I got it. Didn't say it ever worked like the rest of my boards. I used to have an Adafruit S2 in there and that worked. Put your featherS3 in there and it's been non-stop reaching in there to hit the reset button. Can plug it in, run it once, and that works if I don't ever touch it again. That's how I have a pretty picture of it actually working. Try to work with the code in Mu or touch the USB drive to open a file on it and it crashes to the hard fault handler or wifi only works after hard reset (two different problems) at least in my main weatherstation sketch. It's like reaching into a bag of angry cats trying to get it to work right, only works right once after hard reset then crash into hard fault handler if I do anything else. 

It's NOT doing that with the tiny test sketch I just came up with in here, but it's also not connecting to wifi now which is the correlation.  I'm sure if I can get it to connect to wifi it'll crash into the hardfault_handler again. So yeah that's the main issue.

UnexpectedMaker, 2022-10-02:
You wrote "First FeatherS3. Had it in this configuration for months. I love my little weatherstation. Just went through a hurricane with it btw."

Now you say it never worked properly?

I'm so confused and really have no idea where to go to from here.

DJDevon3, 2022-10-02:
It works ONCE. If I touch the code or any file in USB it hard faults forcing a board reset.  So it works... and it doesn't.

DJDevon3, 2022-10-02:
Now with the minimal sketch I've been able to connect reliably, waiting 5 minutes reloading, no errant behavior like I would normally get while it's connected to wifi in the main weatherstation sketch. Seems ok. Does this prove the culprit doesn't exist in minimal sketch? Now going to add adafruit_requests and try some basic GET requests to OpenWeather.

UnexpectedMaker, 2022-10-02:
Remove Mu from all of this. Close it and keep it closed, and just use notepad.

Others have reported inconsistent issues using Mu on Windows. 1 person cant get the code.,py to reload when editing via Mu, and one cant always get the code.py to update on the device using Mu.

So please just use notepad and file explorer to test with.

Also, with WiFi, please ensure the channel you are using is 9 or greater for your testing.

DJDevon3, 2022-10-02:
Set channel=11. Wifi only works after hard reset. If I change the file the wifi doesn't work, will search for wifi forever no ssid. 

Because I have it connected to a TFT I can usually see when it reloads, the screen will flash black, I'll see blinka in the top left corner, etc..  After sitting idle for a couple minutes, if I hit save in notepad++ nothing happens, no reload.

Hit reset, TFT reloads, searches and finds wifi, it will continue functioning normally in this state forever pulling info from OpenWeather every 15 minutes without fail.  

From here there are 2 situations that can happen:
1. Open code.py in notepad++ immediately after hard reset, hit save, no wifi.
2. Open code.py in notepad++, wait 5 minutes, hit save, no reload.

UnexpectedMaker, 2022-10-02:
Can you please work on a board that is not in the case or connected to anything...   Too many variables, I want to focus on the wifi issues and the no reloading on a bare FS3.

UnexpectedMaker, 2022-10-02:
I've had no reports of anything like this in any other dev platform other than CP - and only 3 instances of "code no reloading" as I mentioned above, and only in the last 2-3 weeks... and the FS3 has been out for > 8 months with no HW revisions.

So we need to track down what has changed from a CP point of view (IDF changes under the hood etc) and get some 100% reproducible steps under Windows and then see if they are reproducible on macOS and Linux.

DJDevon3, 2022-10-02:
If I work on a board not connected to a TFT and use notepad++ there's no way to know if it's connecting to wifi, successful GET request, or crashing. Without REPL I'm working blind so I don't see what that's going to accomplish. I can use VScode with serial monitor but I have to reload serial monitor with every reset doing ctrl+d which reloads the code which in turn refuses to connect to wifi because it's not monitoring the first run after a hard reset.

UnexpectedMaker, 2022-10-02:
So many options….
- use the rgb led to show connection status
- Use the blue led on IO13 to show status
- Use ‘tio’ to monitor serial output as it does auto reconnect
- Use mqtt or adafruitio and publish results online

DJDevon3, 2022-10-02:
running this code now, let's call it minimal board only.
```py
import wifi
import socketpool
import board
from digitalio import DigitalInOut, Direction, Pull

# Initialize WiFi Pool
pool = socketpool.SocketPool(wifi.radio)

# Init Blink LED
led13 = DigitalInOut(board.LED)
led13.direction = Direction.OUTPUT

try:
    from secrets import secrets
except ImportError:
    print("Secrets File Import Error")
    raise

# Connect to Wi-Fi
print("\n===============================")
print("Connecting to WiFi...")
while not wifi.radio.ipv4_address:
    try:
        wifi.radio.connect(secrets['ssid'], secrets['password'])
        led13.value = True
    except ConnectionError as e:
        print("Connection Error:", e)

# With the 205 Error it never reaches below this point
print("Connected!\n")
```
Can't use tio in windows. Last time I tried it, it doesn't work and I uninstalled WSL trying to figure this stuff out.

DJDevon3, 2022-10-02:
Installed the new 8.0.0.beta1 released last night. Getting a bunch of different results now.
```py
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.1\]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.1\
===============================
Connecting to WiFi...
Connection Error: Unknown failure 4
Connection Error: Unknown failure 204
Connection Error: Unknown failure 204
Connection Error: Unknown failure 4
Connection Error: Unknown failure 204
Connection Error: Unknown failure 2
Connection Error: Unknown failure 4
Connection Error: Unknown failure 15
Connection Error: Unknown failure 4
Connection Error: Unknown failure 2
Connection Error: Unknown failure 4
Connection Error: Unknown failure 4
Connection Error: Unknown failure 4
Connection Error: Unknown failure 2
Connection Error: Unknown failure 2
Connection Error: Unknown failure 4
```

DJDevon3, 2022-10-02:
It's gotta be something wrong with my code. Just ran the adafruit QT Py S3 internet test code and it works once only after hard reset. https://learn.adafruit.com/adafruit-qt-py-esp32-s3/circuitpython-internet-test

```py
]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.1\Connected to SSID!
My IP address is 192.168.1.2
Ping google.com: 0.000000 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)
----------------------------------------
Fetching json from https://www.adafruit.com/api/quotes.php
----------------------------------------
[{'text': 'We are the music makers, And we are the dreamers of dreams', 'author': "Arthur O'Shaughnessy"}]
----------------------------------------

Fetching and parsing json from https://api.github.com/repos/adafruit/circuitpython
----------------------------------------
CircuitPython GitHub Stars 3187
----------------------------------------
done
]0;🐍192.168.1.2 | Done | 8.0.0-beta.1\
Code done running.
]0;�Wi-Fi: off | Done | 8.0.0-beta.1\Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | Done | 8.0.0-beta.1\
Code stopped by auto-reload. Reloading soon.
]0;�Wi-Fi: off | Done | 8.0.0-beta.1\soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\ESP32-S2 WebClient Test
My MAC addr: ['0xf4', '0x12', '0xfa', '0x42', '0x4', '0x8']
Available WiFi networks:
]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.1\Connecting to Extra 2.4
Traceback (most recent call last):
  File "code.py", line 34, in <module>
]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.1\ConnectionError: No network with that ssid
]0;🐍Wi-Fi: No IP | 34@code.py ConnectionError | 8.0.0-beta.1\
Code done running.
]0;�Wi-Fi: off | Done | 8.0.0-beta.1\
Press any key to enter the REPL. Use CTRL-D to reload.
```
Connects once after hard reset, hit save, auto-reloads and fails to connect.

UnexpectedMaker, 2022-10-02:
Sounds like CP is not handling the cleanup of wifi (or lower level) nicely on a soft reboot... could be related to this...
https://github.com/micropython/micropython/issues/8994

Might also be something else at the firmware level.

Have you tried rolling back to 7.3.0 or 7.2.x (cant remember what version initial S3 support arrived) and seeing if the same thing happens?

UnexpectedMaker, 2022-10-02:
There is no wifi code cleanup at the end of that internet get code - I know the hole "should not need to" - but that assumes a device continuing to run... not a soft reset or even hard reset is reseting the PHY state properly.

What happens if you add `wifi.wifi_radio_set_enabled(False)` at the end of that code?

Neradoc, 2022-10-02:
The autoreload issue is likely #6866 (which I tracked dosn to an ESP-IDF update - see my comment in the thread).

UnexpectedMaker, 2022-10-02:
> The autoreload issue is likely #6866 (which I tracked down to an ESP-IDF update - see my comment in the thread).

Yeah, these issues have all become intertwined and hard to track/ cross reference - well at least for me ;)

anecdata, 2022-10-03:
Some of these errors, namely the ConnectionError unknown errors 2, 4, 15, and 204 seem to me to indicate partial connectivity issues with the AP. What is the RSSI for the AP (same BSSID if multiple APs share an SSID) shown from the device point of view (either `wifi.radio.ap_info.rssi` if connected, or via a wifi scan)? What is the RSSI for the device that's hopefully shown in the GUI of the AP / router?

@DJDevon3  This discussion has gotten quite complex. It may help to summarize at this point your conclusions about which issue(s) are still occurring, whether each is correlated to particular CP version(s), and whether each is correlated to a particular port, board type, or specific board? Are any of the behaviors reproducible with a stripped-down hardware config and simple code... so that others could try it out? With CP 8, is web workflow enabled, with a `/env` file containing wifi credentials? If not, you may want to try some variations with web workflow enabled (it improves wifi scans and some other things, for as-yet mysterious reasons).

@UnexpectedMaker What is the significance of enforcing channel >= 9? I assume we have determined this is not antenna-related?

UnexpectedMaker, 2022-10-03:
@anecdata The ESP32-S3 has a silicon bug when USB is being used with the radio on - there is a 2-3dB loss on lower channels... I Think 6 and below, so I just recommend folks go to 9 or higher if they can.

DJDevon3, 2022-10-03:
https://github.com/espressif/arduino-esp32/issues/2501 found some evidence this is an esp32 interference & module init problem. I have about 3-4 esp32 running at any given time next to each other.  The handshake issue has existed for a while even affecting some magtags which I also have nearby. Also possible my router is having an issue with them. Hard to prove but can't rule out.

DJDevon3, 2022-10-04:
Running S2 internet test found here: https://learn.adafruit.com/adafruit-esp32-s2-feather/circuitpython-internet-test

```py
]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\Connected to UAC 2.4!
My IP address is 192.168.1.215
Ping google.com: 139.000000 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)
----------------------------------------
Fetching json from https://www.adafruit.com/api/quotes.php
----------------------------------------
[{'text': 'As we enjoy great advantages from the inventions of others, we should be glad of an opportunity to serve others by any invention of ours; and this we should do freely and generously.', 'author': 'Benjamin Franklin'}]
----------------------------------------

Fetching and parsing json from https://api.github.com/repos/adafruit/circuitpython
----------------------------------------
CircuitPython GitHub Stars 3190
----------------------------------------
done
]0;🐍192.168.1.215 | Done | 8.0.0-beta.0-72-ga7b10d41b\
Code done running.
]0;�Wi-Fi: off | Done | 8.0.0-beta.0-72-ga7b10d41b\Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
```
OK that's weird. By the time it ends it automatically switches off wifi, is that normal for the test?  

Hit save to auto-reload and it produces this:
```py
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.0-72-ga7b10d41b\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\

ESP32-S2 WebClient Test
My MAC addr: ['0x84', '0xf7', '0x3', '0xd6', '0x8b', '0x72']
Available WiFi networks:
]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\	Extra 2.4		RSSI: -50	Channel: 6
	UAC 2.4		RSSI: -58	Channel: 6
]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;�Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\]
0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\Connecting to UAC 2.4
]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.0-72-ga7b10d41b\Traceback (most recent call last):
  File "code.py", line 31, in <module>
ConnectionError: Unknown failure 2
]0;🐍Wi-Fi: No IP | 31@code.py ConnectionError | 8.0.0-beta.0-72-ga7b10d41b\
Code done running.
]0;�Wi-Fi: off | 31@code.py ConnectionError | 8.0.0-beta.0-72-ga7b10d41b\
Press any key to enter the REPL. Use CTRL-D to reload.
```
This snake formatting has got to go. I cleaned it up a little but REPL output is just a mess now.

anecdata, 2022-10-04:
When code exits, wifi deinits and the connection is dropped.

DJDevon3, 2022-10-04:
Found possible old bug correlation https://github.com/adafruit/circuitpython/issues/3712 Todbot reported exact symptoms I have. Maybe the bug came back?

anecdata, 2022-10-04:
My APs don't have DHCP either. Depending which AP a device connects to, it's path is: device --> one of several APs --> [some paths via one or more of several managed switches] --> router.

anecdata, 2022-10-04:
Until 6866 is resolved, I'd suggest running CP8 as follows:

- run web workflow if possible, even if not needed functionally (I know you said there are issues there)
- avoid wifi scanning if possible, particularly after a reload (reset seems to be OK)
- insert the following before calls to `wifi.radio.connect()`:
```py
wifi.radio.enabled = False
wifi.radio.enabled = True
```

DJDevon3, 2022-10-05:
I’ll have to add some exceptions to my firewall which is a pain to get at and in disarray after moving into a new place. Intend on running Ethernet this winter when my attic isn’t 120F death trap. Might take me a while to get at the firewall stuff.  

I can’t run web workflow wired and especially not wirelessly because port 80 is taken. Melissa’s guide says it doesn’t like running on anything but port 80 right now. That’s a problem for me and most lan admins even if it’s only on a loopback. I’d have to reorganize a good portion of my lan just to test web workflow. That’s beside the point of not being keen on poking holes in my firewall for IOT devices to access my maker pc and the internet.  While I love the idea of web workflow you have to admit it’s going to change the landscape of iot vlans, reconfiguration is required.  I wish it was as simple as just throwing creds into a env file but it’s really not. 

### Summary of this issue to date:
1. Hard reset bug in most cases hides the hard fault bug on the UM FeatherS3. Can’t get it to trigger reliably after updating to latest beta 1 released a few days ago. Progress. 
2. I cannot definitively say the issue is gone because I cannot keep WiFi running in a sketch past the 1st hard reset. Hard reset bug is obscuring hard fault bug. 

Hard fault bug investigation has to be put on hold until the hard reset bug is fixed. Too many variables and getting at it behind the hard reset bug not worth the insanity it’s causing.

anecdata, 2022-10-05:
I infer from the guide that it's just the online Code Editor that hasn't been tested on other ports. The local file browser and code editor should be fine.

But you don't ewven need to use any of those tools to get the connection and scanning benefit. Simply having web workflow enabled seems to improve scanning and connection behavior. Web workflow could be enabled on an alternate port, like 8080, in the `/.env` file, then just forget web workflow exists and do things over USB or standalone as usual.

DJDevon3, 2022-10-12:
The Oct 11th, 2022 nightly build fixed the hard-reset bug on the UM FeatherS3 as well. Pat on the back to DanH & anecdata, and anyone else involved in tracking it down. 

If this hard-fault bug (different than hard-reset bug) does still exist it should get exposed pretty quick in the next couple days. Because I'm using the UM FeatherS3 as my main board in my main long term project if it does hard fault I'll notice it immediately on my weatherstation display. 

I'd like a couple days to put it through paces, should be plenty of time. If all goes well I will gleefully close.

DJDevon3, 2022-10-12:
Woke up this morning to find my weatherstation stopped at 11:30pm last night. Look at the modified timestamp of code.py and it says 10:06pm. It automatically updates every 15 minutes GET data from openweathermaps.  Only had a chance to successfully run about 3 times before crashing. The HardFault bug as I suspected persisted through the fix of the hard-reset bug fix.

```py
]0;🐍Wi-Fi: off | Done | 8.0.0-beta.1-29-gfed884738\Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
]0;🐍Wi-Fi: off | Done | 8.0.0-beta.1-29-gfed884738\
```

DJDevon3, 2022-10-12:
RedBeard in discord suggested a possible correlation between I2C and HardFaults in multiple bug reports across the board. Unplugged I2C BME280, removed code, and set weather update interval to every 3 minutes beginning at 2:02pm.  It's now 2:23pm and no hard faults yet. If it is a memory leak of some kind it might take a while.

DJDevon3, 2022-10-12:
Ran for 2 hours without issue. Tried removing some code to setup an I2C only example and it HardFaulted on hitting save in Mu.  It's got something to do with USB & file save after a certain period of time.

DJDevon3, 2022-10-12:
Ran an I2C test for about an hour, hit save, reloaded without issue.  Issue is not solely I2C related.
```py
I2C BME280 Temp:  89.8
Monotonic 3079.88
I2C BME280 Temp:  89.8
Monotonic 3081.51
I2C BME280 Temp:  89.8
Monotonic 3083.08
```
I created a chart in an effort to convey the behavior on the UM FeatherS3 because it's really hard to explain. There is some kind of inactivity timer involved.
[image]

I don't have an exact time measurement because when it crashes it ejects the USB drive and wipes REPL.  I'd have to try to log REPL it to an SD Card? Unsure if going to that extreme is necessary but I'll do it if anyone thinks it'll help.

DJDevon3, 2022-11-12:
Ported some of this project from the UM Feather S3 + TFT featherwing to an Adafruit Feather S3 4mb/2mb + RGBmatrix panel + RGB matrix featherwing.  It's also crashing into the hard fault handler. Reliably.  Only experimented with it for a couple hours last night. Initial impression is the timing related bug is still present on 8.0.0 beta 4. Fault handler crashes to safe mode which requires a hard reset. It only crashes when wifi is added to the code, was trying pull API data from an online source (openweathermaps).  Using wifi and displayio in both scripts/scenarios on different hardware. Still looking into it. Just an update from a beta testing perspective. Another correlation is after hard reset it will run no problem and trying to make a change and save in Mu triggers the hard fault.

dhalbert, 2022-12-15:
Can you summarize the situations where it crashes?
1. If you start from a hard reset and never edit any files, does it keep running indefinitely?
2. If you start from a hard reset, and edit a file, so that it restarts, does it crash reliably? Immediately or much later?

[4 more replies]

closed 2023-04-15 by DJDevon3