adafruit/circuitpython #6792: i2ctarget.I2Ctarget(board.SCL, board.SDA, [0x41]) fail, must use I2CPeripheral

dglaude, 2022-08-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
import board

#from i2ctarget import I2CTarget
#target = I2Ctarget(board.SDA, board.SCL, [0x41])
# ^^^ FAIL with: File "code.py", line 4, in <module> | NameError: name 'I2Ctarget' is not defined

import i2ctarget
#target = i2ctarget.I2Ctarget(board.SCL, board.SDA, [0x41])
# ^^^ Fail with: File "code.py", line 8, in <module> AttributeError: 'module' object has no attribute 'I2Ctarget'

target = i2ctarget.I2CPeripheral(board.SCL, board.SDA, [0x41])
# ^^^ This is actually working

while True:
    request = target.request()
    if request is not None:
        if request.is_read:
            print("Write request ")
            request.write(b'B')
        else:
            r = request.read()
            if r == "T":
                print("Read a T")
```

### Behavior

Failure message are included in the code.
The code is almost identical to example in #6758 (except for the pin in use), but it fail.
The only way to make it work is to use the "old" naming `target = i2ctarget.I2CPeripheral(board.SCL, board.SDA, [0x41])`.

### Additional information

I did not investigate further on other hardware, so I don't know if this is RP2040 specific.

deshipu, 2022-08-22:
It's `I2CTartget` not `I2Ctarget`, see https://github.com/adafruit/circuitpython/blob/main/shared-bindings/i2ctarget/__init__.c#L98

dglaude, 2022-08-22:
Thanks, I did copy and paste multiple time from #6758 first message.
Or maybe copying the `target` from `i2ctarget` just after an `I2C` and each time I made the case error.
I could not understand as I knew the two were aliases.

This is working (but somehow unstoppable by ^C):
```
import board
from i2ctarget import I2CTarget

target = I2CTarget(board.SCL, board.SDA, [0x41])

while True:
    request = target.request()
    if request is not None:
        if request.is_read:
            print("Write request ")
            request.write(b'B')
        else:
            r = request.read()
            if r == "T":
                print("Read a T")
```

closed 2022-08-22 by dglaude