adafruit/circuitpython #6795: Web Workflow via Wi-Fi Access Point

anecdata, 2022-08-23:
Like #6473, but for Access Point: allow Wi-Fi AP SSID and password to be defined in `.env`, and if present, start up a Wi-Fi AP automatically. If web workflow is enabled, allow web workflow to operate via the AP.

Some discussion of the motivations is here: <https://gist.github.com/anecdata/bddc29b0d21f03c339e8641794addeca>. Particularly useful for external (but reasonably trusted) wifi networks, with non-native-USB CircuitPython devices, when a general-purpose computer with serial isn't readily available.

Rybec, 2024-02-27:
This please!  I don't always have a network I can connect to, so being able to do this would be really handy for working from anywhere away from home.

Rybec, 2024-02-27:
For precedence, maybe have it try to connect to the outside network first, and if that fails make AP mode the fallback?  For my uses, that would be ideal.  When I'm at home, it could just connect to my home network, and when I'm out it would go to AP mode, making it so that I don't have to change settings.toml to get the correct behavior wherever I am.  (Also, this would give anyone using these devices an easy to way to get in and change settings.toml if they suddenly can no longer use the wireless network they have it set for.)

anecdata, 2024-02-27:
More discussion along those lines in 8405.