adafruit/circuitpython #6799: PulseOut not doing 100% duty cycle

HelenFoster, 2022-08-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
import array
import board
import time
import pulseio

pulseOut = pulseio.PulseOut(board.A0, frequency=10000, duty_cycle=0xFFFF)

pulses = array.array('H', [10000])

while(True):
    pulseOut.send(pulses)
    time.sleep(5)
```

### Behavior

Expect a 10ms pulse, but it has short gaps in it every 100us.

### Additional information

Looks like it's this. Since the max value of `duty_cycle` is 0xFFFF, I think the answer for that should be 100%, but it's 99. Should we divide by 0xFFFF instead of `1 << 16`?
https://github.com/adafruit/circuitpython/blob/6572f85913a7ee7f502295568b18b713ff6f1adb/ports/espressif/common-hal/pulseio/PulseOut.c#L48

dhalbert, 2022-08-25:
> Looks like it's this. Since the max value of `duty_cycle` is 0xFFFF, I think the answer for that should be 100%, but it's 99. Should we divide by 0xFFFF instead of `1 << 16`?

That looks suspicious, certainly. Would you like to try a pull request and test the proposed fix? We can help you along with getting a build environment set up (Linux is by far the easiest). If you'd rather not, we're happy to try.

dhalbert, 2022-09-15:
I tried this obvious change, and confirmed the duty cycle is set to 100 internally. However, the waveform does not become uniform.

This is before the change:
[image]

This is after the change (changing divisor to `0xffff`):
[image]

The RMT documentation does not explain this, and I even see recommendations to set the duty cycle to 100 in websearches.

Internally, the percentage is used to set counts that give the time low and time high. In the case of 100%, the time low count goes to zero, but I'm guessing that the peripheral is actually not doing zero counts, but is wrapping around completely.

I don't see complaints about this in the `esp-idf` repo or elsewhere. If you have time to research this a bit further, I would be grateful.

Note that it does not work to set `duty_cycle=0xffff+1` in the Python code, because the argument is passed as a uint16_t, and the top bit gets dropped. It should do a better job of validation.

HelenFoster, 2022-09-19:
Thanks for looking into it. As we were talking about on Discord, I worked through getting the build environment set up, but I haven't really had time since then :(

mwisslead, 2022-10-06:
Maybe just turn the carrier off when duty cycle would be 100?

Also there is a similarly strange effect when duty cycle is set to 0

HelenFoster, 2022-10-07:
If a duty cycle of 100% really doesn't work, that's strange that nobody noticed so far, but from the documentation it looks like disabling the carrier for the 100% case should solve it.

tannewt, 2023-04-14:
@igrr Any ideas?

igrr, 2023-04-14:
I'll check internally. Not sure that the RMT carrier feature supports 0/100% duty cycles in hardware. I suspect the driver is not validating these edge cases. Wiring the pin to constant high or low signal could be a reasonable workaround for the time being.

HelenFoster, 2023-04-14:
0% duty cycle is constant low. But 100% duty cycle isn't constant high, it's a pulse chain without a carrier (which has plenty of real use cases). The RMT should be able to do this by disabling the carrier option, but I haven't tried.

dhalbert, 2024-07-03:
@timchinowsky Do you think this might be addressed by your recent PWM fix?

dhalbert, 2024-07-03:
@timchinowksy Never mind, this is Espressif.